me-no-dev/ESPAsyncWebServer #692: Triggering a redirect asynchronously

deadbird99, 2020-01-30:
I'm facing an issue with redirect. Here is the context.

I've got an HTML page that's served and keeps the user waiting for a process to complete. When the said process is complete, I'd like to serve another page.

To do so, I use a common callback that serves the waiting page and launches the background process.

But when the process completes, I cannot redirect as I don't have any `AsyncWebServerRequest` request to respond to.

Do you guys have an idea on how to circumvent this problem?

IoTThinks, 2020-02-04:
Use AJAX or Jquery to check then load new page if needed.

deadbird99, 2020-02-04:
I thought about that, but ESPAsyncWebServer doesn't handle AJAX requests.

Tofandel, 2020-02-04:
The problem is your waiting page, the request is already done when you serve your page so you can't redirect from the webserver, so either get rid of the waiting page and only send a 302/301 redirect header when the process is done

Or make your waiting page reload itself every 10sec or so and have a flag on the esp to know if the background process is running
If it is continue serving the waiting page if not do the redirection

deadbird99, 2020-02-04:
That's interesting, and you really nailed my problem.

Indeed, when my waiting page has been served, I can't send another response. And as far as I tested, using AJAX requests won't trigger anything on the server side. @Tofandel : if you had this working, could you please share the code?

closed 2020-04-18 by stale[bot]