me-no-dev/ESPAsyncWebServer #693: SPIFFS Server with Authentication ESP8266

wilkerman265, 2020-01-30:
Hello to everyone, im new programming with the ESP8266 (NodeMCU) and im working in a page which can read data from different sensors (DS18B20 and DHT22) and serving it to a web server, all works good by so far, but I need a basic auth so not everyone can access to the page. 

I´ve readyng the ESPAsyncWebServer info and it look like it has a command which it will help me, but I cannot make it work, it enters the page without auth. 

My code part which Im trying to put the auth looks like this 
```
// Print ESP8266 Local IP Address
  Serial.println(WiFi.localIP());
  // Route for root / web page
  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(SPIFFS, "/index.html", String(), false, processor);
  });
```

IoTThinks, 2020-02-04:
Check the homepage and search for "authentication".

There is a link to authenticate for static file.

wilkerman265, 2020-03-01:
I solved my problem editing the  WebAuthentication.cpp like described in this post 
[](https://github.com/me-no-dev/ESPAsyncWebServer/issues/667)

closed 2020-03-01 by wilkerman265