me-no-dev/ESPAsyncWebServer #694: ESPAsyncWebServer with DNSServer does not work

Nexus1212, 2020-02-01:
Hello,
I want du build on a ESP-Wroom-32 a capative portal.

Therefore I want to setup a Accespoint, DNS-Server and Wabserver. 

I decided me to try as a Webserver the ESPAsyncWebServer, because it is very fast and provided a easy api to use web socket.

My Problem:
If any client connected to the Accespoint of the ESP32, the ESP make a restart.

Is it a bug in the ESPAsyncWebServer-Lib?

My Code:

`
#include <WiFi.h>
#include <WiFiClient.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <DNSServer.h>

const char * wifiName = "Testserver";

IPAddress apIP(192, 168, 1, 1);

AsyncWebServer server(80);

// DNS server
const byte DNS_PORT = 53;
DNSServer dnsServer;

void setup() {
  Serial.begin(115200);
  while (!Serial) {
  }
  Serial.println("Setup Start");

 //Wifi

 WiFi.mode(WIFI_AP);
 WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
 WiFi.softAP(wifiName);

 /* Setup the DNS server redirecting all the domains to the apIP */
 dnsServer.setErrorReplyCode(DNSReplyCode::NoError);
 dnsServer.start(DNS_PORT, "*", apIP);
 Serial.println("dnsServer gestartet.");

  server.onNotFound([](AsyncWebServerRequest *request) {

    Serial.println("address");
    Serial.println(request->host());
    Serial.println(request->url());

    request->send(200, "text/plain", "Login Success!");
    
  });

Serial.println("Starte Server");
server.begin();
Serial.println("Setup Ende");
  
}

void loop() {
 //DNS
 dnsServer.processNextRequest();
}
`

The error massage:

`
Setup Start
dnsServer gestartet.
Starte Server
Setup Ende
dhcps: send_offer>>udp_sendto result 0
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060030  A0      : 0x80114dfd  A1      : 0x3ffb3d30  
A2      : 0x3ffcf674  A3      : 0x3ffcf8a8  A4      : 0x3ffced78  A5      : 0x3ffced58  
A6      : 0x0204a8c0  A7      : 0x6504a8c0  A8      : 0x80114ca0  A9      : 0x3ffb3cf0  
A10     : 0x3ffcf684  A11     : 0x3ffcf8a8  A12     : 0x3ffb3d3c  A13     : 0x00000044  
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb3d30 0x40114dfa:0x3ffb3d70 0x40120ae5:0x3ffb3d90 0x40125aed:0x3ffb3dd0 0x4012ad8a:0x3ffb3df0 0x40115697:0x3ffb3e10 0x40088b7d:0x3ffb3e40

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
`

IoTThinks, 2020-02-04:
WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
WiFi.softAP(wifiName);

Put the softAP before softAPConfig.
Or just remove softAPConfig to use default IP 192.168.4.1

You will be able to access EspAsync in 192.168.4.1.

Please note that you wont get the captive portal popup.
But it is captive portal issue.

Nexus1212, 2020-02-05:
Thank you for your response!

That solves my problem. 

Could you explain the capative portal issue please?

IoTThinks, 2020-02-05:
Normally, in firewall or captive portal solution, you need to block the traffic and sniff the http packet then inject a http redirect. 
Then there is popup in phone.

For dnsserver, it resolves every domain names to the esp IP only.
It is not enough for iOS and Android to popup.
You need to open browser and key in any http website.

I will investigate more.
But clearly, it's the issue with captive portal code than the esp async web server.

Nexus1212, 2020-02-05:
I allready found that hind. Thank you! My current solution is:

  server.onNotFound([](AsyncWebServerRequest *request) {
    AsyncWebServerResponse *response = request->beginResponse(302); //Sends 302 Weiterleitung
    response->addHeader("Location", "http://myportal/login");
    request->send(response);
  });

 server.on("/login", HTTP_GET, [](AsyncWebServerRequest *request){
  request->send(200, "text/html", PageLogin);

  It works well!
  but sometimes the esp32 does over 10s nothing or the esp restarts... is there another thing?

IoTThinks, 2020-02-05:
Nice idea!
Let me try later and let you know. :+1:

IoTThinks, 2020-02-05:
Your code works great on iPhone!
However, it does not show popup in my Android.

Let me dig and try further.
Thanks for your awesome code.

IoTThinks, 2020-02-05:
iOS12 tries to get http://captive.apple.com/hotspot-detect.html. Hence, it reaches notFoundHandler. And popup as expected.

Android 9 has no attempt to load url.
May be it loads something like http://google... but not http://google.../something.html.
And we have dnsServer.start(DNS_PORT, "*", WiFi.softAPIP()); ==> Map google domain to our own IP.
Hence, Android may think the url is loaded OK. So no popup.

CircuitSetup, 2020-03-28:
I was having a very similar issue, but never found a good solution. Was this every resolved?

closed 2020-08-09 by stale[bot]