me-no-dev/ESPAsyncWebServer #695: ws.print() doesn't work

sfranzyshen, 2020-02-02:
I'm trying to use `ws.printf(id, "%s", cmdString.c_str());`
but nothing is being received at the client ... it works fine when I replace ws.printf() function with the ws.printfAll() function ... i'm able to use client->printf() while in the event handler ...

```
AsyncWebSocket ws("/ws");
void processCommand(String cmdString, uint32_t id);
processCommand(msg, client->id());
ws.printf(id, "%s", cmdString.c_str());      <-- Doesn't work
ws.printfAll("%s", cmdString.c_str());       <-- Works
```
am I missing something?

sfranzyshen, 2020-02-03:
Has this been a problem since 2016? #56 ... If the problem wasn't addressed or fixed ... then why the hell does the issue get closed and forgotten? Stop wasting peoples time ... at very least make a post in the documentation that there is a problem and that this function is borked!!!

sfranzyshen, 2020-02-03:
stale bot is not useful ... it degrades development ... please eliminate this feature from this repository ...

sfranzyshen, 2020-02-03:
... I think that this problem has been "can kicked" for a long time ... and now ... I too have changed direction and have eliminated my need for the ws.printf() function ... so now this too is the end of my ranting about the problem. It would be nice to once and for all fix this ... but since it's not a issue to the developers (and now not an issue to me) I guess it'll linger on and on ... I just hope that someone else trying this out can find these post (if stale bot gets to them) and avoid hours of aggravation ...

BlueAndi, 2020-02-03:
https://github.com/me-no-dev/ESPAsyncWebServer/blob/f13685ee97675be2ac9502d177d3024ebc49c1e0/src/AsyncWebSocket.cpp#L1014-L1024

It looks like the problem is that the client printf() method is called, but the client should provide a vprintf() method which can deal with the variable argument list.

BlueAndi, 2020-02-04:
@sfranzyshen Can you confirm that the bugfix works for you?

sfranzyshen, 2020-02-05:
> @sfranzyshen Can you confirm that the bugfix works for you?

Yes! Yes I can Thank You very much for putting this to rest ...

closed 2020-02-05 by sfranzyshen