me-no-dev/ESPAsyncWebServer #699: How to run ESPAsyncWebServer with 2 Different IPs?

rainG92, 2020-02-06:
I am using ESP8266.
I want to run ESP8266 in STA+AP (Station Mode + Access Point Mode) at once.
IP address of Station Mode is dynamic IP to suit devices with different IP Pool ranges.
IP address of Access Point Mode is set to be static IP.

I created simple wifi manager page that meant to be accessible from either Station Mode or Access Point Mode.
Thus IP Address of Access Point Mode must be fixed, to ensure user can always reconfigure WIFI through web page hosted on ESP8266.

In my sketch, Access Point Mode will be set up first and followed by Station Mode.
However I am faced with a problem by having 2 Different IPs. (It works fine if both IPs are the same, I can access webpage via Station Mode or Access Point Mode.)

For example: 
My Access Point Mode IP : 192.168.1.1
My Station Mode IP : 192.168.1.2
Web Page is hosted as root

If i connected via Access Point Mode, i can access webpage through 192.168.1.1.
If i connected via Station Mode, i can not access webpage through 192.168.1.1 nor 192.168.1.12

Sketch:
AsyncWebServer server(80);

Is there any methods for me to add another AsyncWebServer with IP Address?
For example : 
AsyncWebServer server2(WiFi.localIP(),80);

My references :
https://arduino.stackexchange.com/questions/67269/create-one-server-in-ap-mode-and-another-in-station-mode

closed 2020-02-12 by rainG92

shwetankshrey, 2020-03-04:
Hi @rainG92 !
Were you able to solve this?

rainG92, 2020-03-06:
@shwetankshrey Yes I did,  At first i did try to create such function myself via referencing to ESPAsyncTCP library.  But i discovered the root problem by chance, I realized the main problem is the Access Point Mode Static IP configuration. There seems to be a limitation or range of valid IP address for esp8266, but i couldn't find any documentation regarding this limitation. My best bet is is try and trial for valid static IP address which is within the valid range. You can know whether which IP address is valid by observing serial terminal logging.  

You can refer the code at here https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/soft-access-point-class.html

Note : Failed WIFI SOFTAP configuration means invalid IP and still make you encounter similar problem as I did. Just try your luck with another set of IP address.

Sorry for the elaborated description as I think  this will help those who want to dive in further in related issues.