me-no-dev/ESPAsyncWebServer #701: Form validation class?

nardev, 2020-02-11:
As there is more and more http rest style of communication, it's pretty logical to have a class for basic validations in C++.

I tried to find something, but no luck so far. It's ok to implement it from time to time, copy/paste some functions but it would be much better maintained and probably safer if there was an opensource class, openly maintained, where bugs could be tracked and fixed.

**Did i miss something like that already?**

For example, i would like to be to easily validate:

- ip address
- url
- integer range
- length
- email
- containing numbers and letters
- number/string from an existing array 

and, maybe someone else have more ideas.

**Any thoughts?**

Pablo2048, 2020-02-11:
Hi, you can use html5 validators in your html/js code. I think that validating form input at the server side of web application is depreceated - in case of wrong input there is big overhead to bring everything back to the client and highlight errors IMHO.

nardev, 2020-02-11:
Well, if you don't have web interface but a rest API? Or if you decide to
send json formated config file with a lot of data via POST?

Pablo2048, 2020-02-11:
If you do so this way, you have to write separate library for doing this. It has nothing to do with the ESPAsyncWebServer library.

nardev, 2020-02-11:
Well, you are right of course. But where else can i initiate such conversation? Isn't this the perfect place :)

Pablo2048, 2020-02-11:
:-) I suggest to try esp8266.com , or reddit, or... I did an quick search for "parameter validation library c++" and found this https://www.learncpp.com/cpp-tutorial/185-stream-states-and-input-validation/ so it seems like regex implementation is the best way to do it :D

nardev, 2020-02-11:
mmm i think that you didn't understand my point... sorry

Pablo2048, 2020-02-11:
Oh, so sorry...

closed 2020-04-25 by stale[bot]