me-no-dev/ESPAsyncWebServer #704: Help with compile errors

esp32noob, 2020-02-14:
ESPAsyncWebServer is a fantastic product that I use in numerous projects. Up until yesterday I had no problems with compiling it.  Now any project that I compile with it, existing or new, I get the errors below.

I use:
Microsoft Windows 10 Insider Preview Version 2004 Build 19564.1005
Microsoft Visual Studio version 1.43.0-insider.
Platformio version 1.10.0.
Espressif 32 version 1.11.2.
AsyncTCP version 1.1.1.
ESP Async WebServer version 1.2.3
ESPAsyncTCP version 1.2.2.

I'm looking for guidance as to how to proceed in fault finding, it must be something that has changed in my environment but I can't think what other than a Windows update that happened last night.  I did my last successful compile at 11pm last night, left my pc on overnight and can't compile this morning.

I have unistalled all libraries, uninstalled PlatformIO, deleted the PlatformIO directory, rebooted, reinstalled PlatformIO and reinstalled the libraries. This solved the problem of not being able to find the interrupts.h file which had also appeared but not the following errors. I have also searched the web and reviewed the issues on this site.

This is the new app I'm trying to compile:
#include <Arduino.h>
#include <ESPAsyncWebServer.h>
AsyncWebServer WebServer(80);
void setup()
{
  Serial.begin(115200);
  Serial.println("Async web server test");
  WebServer.on("/rdata", HTTP_GET, [](AsyncWebServerRequest *request) {
    request->send(200, "text/html", "Loop text");
  });
  // Catch all
  WebServer.onNotFound([](AsyncWebServerRequest *request) {
    request->send(200, "text/html", "Default text");
  });
  WebServer.begin();
}
void loop()
{
  vTaskDelay(pdMS_TO_TICKS(2000));
}
But it does not compile.  I get the same errors when compiling existing working apps.

The errors:
Compiling .pio\build\esp32doit-devkit-v1\lib935\ESPAsyncTCP_ID305\ESPAsyncTCPbuffer.cpp.o
In file included from C:\Users\Terminal\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\lwip/lwip/opt.h:51:0,
                 from C:\Users\Terminal\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\lwip/lwip/init.h:40,
                 from C:\Users\Terminal\.platformio\lib\ESPAsyncTCP_ID305\src\ESPAsyncTCP.h:31,
                 from C:\Users\Terminal\.platformio\lib\ESPAsyncTCP_ID305\src\AsyncPrinter.h:26,
                 from C:\Users\Terminal\.platformio\lib\ESPAsyncTCP_ID305\src\AsyncPrinter.cpp:22:
C:\Users\Terminal\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\lwip/lwipopts.h:326:0: warning: "TCP_MSS" redefined
 #define TCP_MSS                         CONFIG_TCP_MSS
 ^
In file included from C:\Users\Terminal\.platformio\lib\ESPAsyncTCP_ID305\src\ESPAsyncTCP.h:25:0,
                 from C:\Users\Terminal\.platformio\lib\ESPAsyncTCP_ID305\src\AsyncPrinter.h:26,
                 from C:\Users\Terminal\.platformio\lib\ESPAsyncTCP_ID305\src\AsyncPrinter.cpp:22:
C:\Users\Terminal\.platformio\lib\ESPAsyncTCP_ID305\src/async_config.h:13:0: note: this is the location of the previous definition
 #define TCP_MSS (1460)
 ^
Plus 2 more similar errors
-----------------------
C:\Users\Terminal\.platformio\lib\ESPAsyncTCP_ID305\src\AsyncPrinter.cpp: In constructor 'AsyncPrinter::AsyncPrinter(AsyncClient*, size_t)':
C:\Users\Terminal\.platformio\lib\ESPAsyncTCP_ID305\src\AsyncPrinter.cpp:48:11: error: 'panic' was not declared in this scope
     panic(); //What should we do?
Plus 8 more similar errors
-----------------------
C:\Users\Terminal\.platformio\lib\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp: In member function 'bool AsyncClient::connect(IPAddress, uint16_t)':
C:\Users\Terminal\.platformio\lib\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:248:8: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
   addr.addr = ip;
        ^
Plus 6 more similar errors
-----------------------
*** [.pio\build\esp32doit-devkit-v1\lib935\ESPAsyncTCP_ID305\AsyncPrinter.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\lib935\ESPAsyncTCP_ID305\ESPAsyncTCPbuffer.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\lib935\ESPAsyncTCP_ID305\ESPAsyncTCP.cpp.o] Error 1

Any help would be appreciated.

eyer1951, 2020-02-14:
Same problem. Compiled fine yesterday, these same errors today. Platformio offered an update this morning, which I accepted. Problems after that. 
```
Dependency Graph
|-- <ESP Async WebServer> 1.2.3
|   |-- <ESPAsyncTCP> 1.2.2
|   |-- <AsyncTCP> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0

```

brunohorta82, 2020-02-14:
same problem where,  :(

brunohorta82, 2020-02-14:
change ESPAsyncTCP  to -> AsyncTCP if you use ESP32 boards

esp32noob, 2020-02-14:
Thanks for your quick response.  You're talking to a noob here, how do I go about changing this?  I don't specify either in my header entries, I assume that it is call by ESPAsyncWebServer?

BlueAndi, 2020-02-14:
Add
```
lib_compat_mode=strict
lib_ldf_mode=chain+
```
to your platformio.ini
and build again. Maybe you have to remove .pio directory first ... but try.

esp32noob, 2020-02-14:
Thank you.  Fixed.  I uninstalled the ESPAsyncTCP library, now I can compile again.  I assume I'll have to watch for this after any update.

closed 2020-02-14 by esp32noob

eyer1951, 2020-02-15:
After adding these two lines
```
lib_compat_mode=strict
lib_ldf_mode=chain+
```
 to platformio.io, I get this error:
```
In file included from C:/Users/Mark/.platformio/lib/WebSockets_ID549/src/WebSocketsServer.h:28:0,
                 from src\main.cpp:68:
C:/Users/Mark/.platformio/lib/WebSockets_ID549/src/WebSockets.h:166:30: fatal error: WiFiClientSecure.h: No such file or directory
```

I am using lots of includes, including
```
#include <WebServer.h>
#include <WiFi.h>
#include <ESPmDNS.h>
#include <ArduinoOTA.h>
#include <WebSocketsServer.h>
```
Did not try removing .pio directory yet. Will try.

eyer1951, 2020-02-15:
If I have the same exact problem that another person marks as "closed," yet my problem remains, is it appropriate to copy it and open it as a new issue? Or, is it possible (or better) to re-open a closed issue?

proddy, 2020-02-15:
This is due to a bug in the recent PlartformIO 4.2.0 where the LDF is not using the transitive dependencies defined in libray.json files and compiling everything.

A quick fix is simply to add `lib_ignore = AsyncTCP` to the [env] section of your `platformio.ini` file if using ESP8266 or `lib_ignore = ESPAsyncTCP` if building for the ESP32.

@eyer1951 I don't think it has been logged as a bug in pio yet, but someone will very soon I suspect.

MohammedNoureldin, 2021-03-08:
@eyer1951 , have you managed to get it compiling with WebSocket and `lib_ldf_mode=chain+`? It still not works for me.

eyer1951, 2021-03-08:
@MohammedNoureldin, when I updated PlatformIO (now at core 5.1.0), that problem went away.  No special lib_ldf_mode needed.

MohammedNoureldin, 2021-03-08:
@eyer1951 so you are using the default implicit value `chain`?

eyer1951, 2021-03-08:
Yes