me-no-dev/ESPAsyncWebServer #705: "TCP_MSS" redefined; interrupts.h: No such file or directory

m-menon, 2020-02-15:
hi, when I tried to compile a project on PlatformIO (espressif / Arduino framework ), I came across this error. It might also have something to do with ESP Async TCP but I thought that I'd post it here because ESPAsyncWebServer needs the async tcp library to work.

```
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.11.2 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.200129 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP Async WebServer> 1.2.3
|   |-- <ESPAsyncTCP> 1.2.2
|   |-- <AsyncTCP> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.2
|-- <AsyncTCP> 1.1.1
|-- <WiFi> 1.0
Building in release mode
Compiling .pio\build\esp32dev\src\main.cpp.o
Generating partitions .pio\build\esp32dev\partitions.bin
Compiling .pio\build\esp32dev\libdd7\ESPAsyncTCP_ID305\AsyncPrinter.cpp.o
Compiling .pio\build\esp32dev\libdd7\ESPAsyncTCP_ID305\ESPAsyncTCP.cpp.o
Compiling .pio\build\esp32dev\libdd7\ESPAsyncTCP_ID305\ESPAsyncTCPbuffer.cpp.o
Compiling .pio\build\esp32dev\libdd7\ESPAsyncTCP_ID305\SyncClient.cpp.o
Compiling .pio\build\esp32dev\libdd7\ESPAsyncTCP_ID305\tcp_axtls.c.o
Compiling .pio\build\esp32dev\lib065\AsyncTCP_ID1826\AsyncTCP.cpp.o
In file included from C:\Users\RW\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\lwip/lwip/opt.h:51:0,
                 from C:\Users\RW\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\lwip/lwip/init.h:40,
                 from .pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\ESPAsyncTCP.h:31,
                 from .pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\AsyncPrinter.h:26,
                 from .pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\AsyncPrinter.cpp:22:
C:\Users\RW\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\lwip/lwipopts.h:326:0: warning: "TCP_MSS" redefined
 #define TCP_MSS                         CONFIG_TCP_MSS
 ^
In file included from .pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\ESPAsyncTCP.h:25:0,
                 from .pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\AsyncPrinter.h:26,
                 from .pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\AsyncPrinter.cpp:22:
.pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src/async_config.h:13:0: note: this is the location of the previous definition
 #define TCP_MSS (1460)
 ^
Compiling .pio\build\esp32dev\lib6ab\FS\FS.cpp.o
Compiling .pio\build\esp32dev\lib6ab\FS\vfs_api.cpp.o
In file included from C:\Users\RW\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\lwip/lwip/opt.h:51:0,
                 from C:\Users\RW\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\lwip/lwip/init.h:40,
                 from .pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\ESPAsyncTCP.h:31,
                 from .pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:75:
C:\Users\RW\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\lwip/lwipopts.h:326:0: warning: "TCP_MSS" redefined
 #define TCP_MSS                         CONFIG_TCP_MSS
 ^
In file included from .pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\ESPAsyncTCP.h:25:0,
                 from .pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:75:
.pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src/async_config.h:13:0: note: this is the location of the previous definition
 #define TCP_MSS (1460)
 ^
In file included from C:\Users\RW\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\lwip/lwip/opt.h:51:0,
                 from C:\Users\RW\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\lwip/lwip/init.h:40,
                 from .pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\ESPAsyncTCP.h:31,
                 from .pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\ESPAsyncTCPbuffer.h:37,
                 from .pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\ESPAsyncTCPbuffer.cpp:29:
[... 83 lines cut ...]
     connect(IPAddress(ipaddr->addr), _connect_port);
                               ^
.pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\ESPAsyncTCPbuffer.cpp: In member function 'size_t AsyncTCPbuffer::_handleRxBuffer(uint8_t*, size_t)':
.pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\ESPAsyncTCPbuffer.cpp:469:21: error: 'panic' was not declared in this scope
               panic(); //TODO: What action should this be ?
                     ^
.pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp: In member function 'uint32_t AsyncClient::getRemoteAddress()':
.pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:837:26: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
   return _pcb->remote_ip.addr;
                          ^
.pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp: In member function 'uint32_t AsyncClient::getLocalAddress()':
.pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:849:25: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
   return _pcb->local_ip.addr;
                         ^
.pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp: In member function 'void AsyncServer::begin()':
.pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:1122:14: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
   local_addr.addr = (uint32_t) _addr;
              ^
*** [.pio\build\esp32dev\libdd7\ESPAsyncTCP_ID305\ESPAsyncTCPbuffer.cpp.o] Error 1
*** [.pio\build\esp32dev\libdd7\ESPAsyncTCP_ID305\ESPAsyncTCP.cpp.o] Error 1
============================================================================ [FAILED] Took 5.88 seconds ============================================================================
The terminal process terminated with exit code: 1
```

Any leads that could help sort this out? Thanks

m-menon, 2020-02-15:
This was the problem; For ESP32 boards we got to ignore ESPAsyncTCP (in PlatformIO) by using `lib_ignore`. The ESP Async Web Server uses AsyncTCP which is the correct lib for ESP32

```
lib_ignore =
  ESPAsyncTCP
```

closed 2020-02-15 by m-menon

atanisoft, 2020-02-15:
@m-menon Another (possibly better) approach is add this to your platformio.ini file:
```
lib_compat_mode=strict
```
With this flag LDF ignores libraries which do not apply to the current platform (among other checks).

m-menon, 2020-02-15:
@atanisoft  okay, cool. 

Thanks for the suggestion - I'll try that out.