me-no-dev/ESPAsyncWebServer #706: Reboot on web request doing extended file activity

esp32noob, 2020-02-15:
>I log data to several files.  When I get a web request I combine info from those files and return the file.  If I have any file transactions longer then 5 seconds the ESP32 reboots, see error below.  Also below is code that reproduces the error.

>This is obviously caused by the extended file activity time.  Based on the error message I assume (a tenuous assumption) that the web server times out and causes a reboot. My question is there any way to extend the wait for the file activity to finish before the reboot.

>Any assistance would be gratefully received.
--------------------------------
**I use:**
>Microsoft Windows 10 Insider Preview Version 2004 Build 19564.1005
Microsoft Visual Studio version 1.43.0-insider.
Platformio version 1.10.0.
Espressif 32 version 1.11.2.
AsyncTCP version 1.1.1.
ESP Async WebServer version 1.2.3.
I don't know how to get the SPIFFS version.
esp32doit-devkit-v1.
--------------------------------
**Error:**
>Async web server file activity test
.Web print request
.Write time: 3882
Read time loop 0: 668
E (14510) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (14510) task_wdt:  - async_tcp (CPU 0/1)
E (14510) task_wdt: Tasks currently running:
E (14510) task_wdt: CPU 0: IDLE0
E (14510) task_wdt: CPU 1: IDLE1
E (14510) task_wdt: Aborting.
abort() was called at PC 0x400dffaf on core 0

>Backtrace: 0x4008c434:0x3ffbe170 0x4008c665:0x3ffbe190 0x400dffaf:0x3ffbe1b0 0x40084771:0x3ffbe1d0 0x4016fc53:0x3ffbc100 0x400e1366:0x3ffbc120 0x4008a361:0x3ffbc140 0x40088b7d:0x3ffbc160

>Rebooting...
-------------------------
**Code to reproduce the issue:**
>#include <Arduino.h>
#include <SPIFFS.h>
#include <WiFi.h>
#include <ESPAsyncWebServer.h>
AsyncWebServer WebServer(80);
void PrintDelay()
{
  // Create a set of data files
  uint32_t LoopTime = millis();
  for (uint8_t k = 0; k < 25; k++)
  {
    File TestFile = SPIFFS.open("/testfile" + String(k) + ".txt", "w");
    // Write text to the files
    for (uint8_t i = 0; i < 100; i++)
    {
      TestFile.println("TestData - " + String(k) + ":" + String(i));
    }
    TestFile.close();
  }
  Serial.println("Write time: " + String(millis() - LoopTime));
  // Read the data files
  for (int i = 0; i < 3; i++)
  {
    LoopTime = millis();
    for (uint8_t k = 0; k < 25; k++)
    {
      File TestFile = SPIFFS.open("/testfile" + String(k) + ".txt", "r");
      while (TestFile.position() < TestFile.size())
      {
        String TestData = TestFile.readString();
      }
      TestFile.close();
    }
    Serial.println("Read time loop " + String(i) + ": " + String(millis() - LoopTime));
  }
}

>void setup()
{
  Serial.begin(115200);
  Serial.println("Async web server file activity test");
  SPIFFS.begin();
  while (!WiFi.begin("SSID", "SSIDPassword"))
    vTaskDelay(pdMS_TO_TICKS(250));
  WebServer.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
    request->send(200, "text/html", "Main page, to test delay use /print");
  });
  WebServer.on("/print", HTTP_GET, [](AsyncWebServerRequest *request) {
    Serial.println("Web print request");
    PrintDelay();
    request->send(200, "text/html", "Print delay completed, use / or /print");
  });
  // Catch all
  WebServer.onNotFound([](AsyncWebServerRequest *request) {
    request->send(200, "text/html", "use / or /print");
  });
  WebServer.begin();
}
void loop()
{
  Serial.print(".");
  vTaskDelay(pdMS_TO_TICKS(10000));
}

BlueAndi, 2020-02-16:
The task needs to much time, because SPIFFS access is very slow.
Try to disable the task watchdog and pin task to core 1.
```
CONFIG_ASYNC_TCP_RUNNING_CORE 1
CONFIG_ASYNC_TCP_USE_WDT 0
```
This must be configured before you include the headers or e.g. in the platformio.ini

esp32noob, 2020-02-19:
Thank you, this has solved my immediate problem.  With the changes the server now responds correctly with the file, I really appreciate your help.  I note that if the file activity takes more than 15 seconds ESPAsyncWebServer becomes unresponsive and needs a reboot to come back.  Is there a way that ESPAsyncWebServer can fail gracefully so that I can recover programmatically?

Thanks

BlueAndi, 2020-02-20:
Hmm ... it should not stop to answer requests. BTW is it possible to add a delay() in your file handling? This would give the server a better chance.
Can you link your code?

esp32noob, 2020-02-21:
There seems to be 2 issues. Using the code above but change (int i = 0; i < 3; i++) to (int i = 0; i < 20; i++).  The 1st issue is that once /print is running in 1 browser tab, I can not connect using another browser tab.  2nd, if you watch the terminal when you do a /print, the app responds correctly and counts the loops then drops into idle mode, printing '.'s on the terminal screen but the web page never returns to the browser, the browser tab sits waiting until it times out.  I realise that a count of 20 is way to long for a normal app but this illustrates the issue. Thanks for your help.

esp32noob, 2020-02-21:
Putting a delay - vTaskDelay(pdMS_TO_TICKS(50)); - before - TestFile.close(); - causes the problem to occur early in the count, i.e. it seems to be time related rather than activity related.

BlueAndi, 2020-02-22:
Its hard to see in the unformatted code block above. ;-)

To 1) Thats right and it behaves as you implemented it! As long as you not return from handling the "GET /print" access, you can't access anything else from side. Because you block the whole task, where the web server is running. The web server can only handle one request at a time. Of course other requests might be queued, but the browser has a timeout as well.

If you try to trigger via web request heavy work in the filesystem, you should consider doing the job in a separate task or in the main loop.

esp32noob, 2020-02-22:
I've figured out how to improve the look of the code above so it should be easier to cut and paste but really all I'm doing is making file activity take a long time.  Before I made the changes you suggested above, the web server would reboot if the file activity took longer the 5 seconds.  After the changes my my app works fine but I noticed that anything over 15 seconds and the web server became unresponsive and never returned the web page.  If you make the loop in the code above longer by changing the loop count from 3 to 20, in the terminal you will see the file activity complete but the web page is not returned and the web server won't respond to further requests.  Your comment above is that this non-resonse is correct as the web server still thinks that it is servicing a request. The question is why does the web server still think that it is busy after the file activity completes.  I would have expected it to the have timed out while waiting for the file activity and returned a 404 or waited until the file activity completed and sent the page.  I appreciate your help with this, thanks.

BlueAndi, 2020-02-23:
Hmm ... but it completes here:
[image]

BlueAndi, 2020-02-23:
I will increase the number of read cycles ...

BlueAndi, 2020-02-23:
Now I can confirm it. I debugged it down till AsyncTcp::_tcp_write(). This is called, but _tcp_write_api() via tcpip_api_call() is never called, as it should. But it seems there is error handling missing.

BlueAndi, 2020-02-23:
Further debugging shows that the _async_queue, which receives lwIP event packets is full after some time. This is caused, because of blocking the task. ;-)

The queue is filled op with LWIP_TCP_POLL events.

As a quick hack, I increased the poll interval from 1 to 10 in the AsyncClient::AsyncClient() and voila the browser receives the response again.

[image]

BlueAndi, 2020-02-23:
I wouldn't consider this as an error, because the task watchdog was original activated to ensure that this case doesn't happen, I guess.

esp32noob, 2020-02-23:
I'm sorry but I don't have the knowledge to be able to comment on your change, I'm only capable of using and testing it however that would explain the behaviour.  Thank you for looking at this, I really do appreciate it. Do you have a version that I can play with or can you do an update?

BlueAndi, 2020-02-23:
Search in your platformio folder in the user profile for the file AsyncTcp.cpp
There you will find the mentioned function, which to change.

esp32noob, 2020-02-28:
Works for me, I have tested it with file activity of 50 seconds and it still responds.  I do need to have the original setting changes as suggested above.  Thanks for your help.

closed 2020-02-28 by esp32noob

meditant, 2020-03-01:
I confirm it's solve a "buzzer" problems for me !

If the buzzer is too long, it's reboot !
Now with this modification it's work fine.

Best