me-no-dev/ESPAsyncWebServer #707: library doesn't compile anymore after new checkout

0xFEEDC0DE64, 2020-02-16:
Hi,
I removed .platformio and .pio from my project, so toolchains and libraries were checked out again. Since that the webserver lib does not compile anymore. Is there some dependency lib version pinning useful?

```
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.11.2 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson> 6.14.1
|-- <ESP Async WebServer> 1.2.3
|   |-- <ESPAsyncTCP> 1.2.2
|   |-- <AsyncTCP> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 6.14.1
|-- <BluetoothSerial> 1.0
|-- <SPI> 1.0
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <TFT_eSPI> 2.0.4
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <FS> 1.0
|   |-- <SPI> 1.0
|-- <WiFi> 1.0
Building in release mode
Compiling .pio/build/esp32dev/lib9fa/ESPAsyncTCP_ID305/AsyncPrinter.cpp.o
Compiling .pio/build/esp32dev/lib9fa/ESPAsyncTCP_ID305/ESPAsyncTCP.cpp.o
Compiling .pio/build/esp32dev/lib9fa/ESPAsyncTCP_ID305/ESPAsyncTCPbuffer.cpp.o
Compiling .pio/build/esp32dev/lib9fa/ESPAsyncTCP_ID305/SyncClient.cpp.o
In file included from /home/feedc0de/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/lwip/lwip/opt.h:51:0,
                 from /home/feedc0de/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/lwip/lwip/init.h:40,
                 from .pio/libdeps/esp32dev/ESPAsyncTCP_ID305/src/ESPAsyncTCP.h:31,
                 from .pio/libdeps/esp32dev/ESPAsyncTCP_ID305/src/AsyncPrinter.h:26,
                 from .pio/libdeps/esp32dev/ESPAsyncTCP_ID305/src/AsyncPrinter.cpp:22:
/home/feedc0de/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/lwip/lwipopts.h:326:0: warning: "TCP_MSS" redefined
 #define TCP_MSS                         CONFIG_TCP_MSS
 ^
In file included from .pio/libdeps/esp32dev/ESPAsyncTCP_ID305/src/ESPAsyncTCP.h:25:0,
                 from .pio/libdeps/esp32dev/ESPAsyncTCP_ID305/src/AsyncPrinter.h:26,
                 from .pio/libdeps/esp32dev/ESPAsyncTCP_ID305/src/AsyncPrinter.cpp:22:
.pio/libdeps/esp32dev/ESPAsyncTCP_ID305/src/async_config.h:13:0: note: this is the location of the previous definition
 #define TCP_MSS (1460)
 ^
In file included from /home/feedc0de/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/lwip/lwip/opt.h:51:0,
                 from /home/feedc0de/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/lwip/lwip/init.h:40,
                 from .pio/libdeps/esp32dev/ESPAsyncTCP_ID305/src/ESPAsyncTCP.h:31,
                 from .pio/libdeps/esp32dev/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:75:
/home/feedc0de/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/lwip/lwipopts.h:326:0: warning: "TCP_MSS" redefined
 #define TCP_MSS                         CONFIG_TCP_MSS
 ^
In file included from .pio/libdeps/esp32dev/ESPAsyncTCP_ID305/src/ESPAsyncTCP.h:25:0,
                 from .pio/libdeps/esp32dev/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:75:
.pio/libdeps/esp32dev/ESPAsyncTCP_ID305/src/async_config.h:13:0: note: this is the location of the previous definition
 #define TCP_MSS (1460)
 ^
In file included from /home/feedc0de/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/lwip/lwip/opt.h:51:0,
[... 88 lines cut ...]
.pio/libdeps/esp32dev/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp: In member function 'uint32_t AsyncClient::getLocalAddress()':
.pio/libdeps/esp32dev/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:849:25: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
   return _pcb->local_ip.addr;
                         ^
.pio/libdeps/esp32dev/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp: In member function 'void AsyncServer::begin()':
.pio/libdeps/esp32dev/ESPAsyncTCP_ID305/src/ESPAsyncTCP.cpp:1122:14: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
   local_addr.addr = (uint32_t) _addr;
              ^
*** [.pio/build/esp32dev/lib9fa/ESPAsyncTCP_ID305/AsyncPrinter.cpp.o] Error 1
.pio/libdeps/esp32dev/ESPAsyncTCP_ID305/src/ESPAsyncTCPbuffer.cpp: In constructor 'AsyncTCPbuffer::AsyncTCPbuffer(AsyncClient*)':
.pio/libdeps/esp32dev/ESPAsyncTCP_ID305/src/ESPAsyncTCPbuffer.cpp:35:15: error: 'panic' was not declared in this scope
         panic();
               ^
.pio/libdeps/esp32dev/ESPAsyncTCP_ID305/src/ESPAsyncTCPbuffer.cpp: In member function 'virtual size_t AsyncTCPbuffer::write(const uint8_t*, size_t)':
.pio/libdeps/esp32dev/ESPAsyncTCP_ID305/src/ESPAsyncTCPbuffer.cpp:127:23: error: 'panic' was not declared in this scope
                 panic();
                       ^
.pio/libdeps/esp32dev/ESPAsyncTCP_ID305/src/ESPAsyncTCPbuffer.cpp: In member function 'size_t AsyncTCPbuffer::_handleRxBuffer(uint8_t*, size_t)':
.pio/libdeps/esp32dev/ESPAsyncTCP_ID305/src/ESPAsyncTCPbuffer.cpp:469:21: error: 'panic' was not declared in this scope
               panic(); //TODO: What action should this be ?
                     ^
*** [.pio/build/esp32dev/lib9fa/ESPAsyncTCP_ID305/ESPAsyncTCP.cpp.o] Error 1
*** [.pio/build/esp32dev/lib9fa/ESPAsyncTCP_ID305/ESPAsyncTCPbuffer.cpp.o] Error 1
========================== [FAILED] Took 2.50 seconds ==========================
```

Pablo2048, 2020-02-16:
Hi,
try to use LDF mode: strict .

0xFEEDC0DE64, 2020-02-16:
@Pablo2048 thanks for the hint, that worked.
For future reference, I added this line to the .ini file:

    lib_compat_mode = strict

closed 2020-02-16 by 0xFEEDC0DE64