me-no-dev/ESPAsyncWebServer #708: calling addHandler on a running webserver throws an exception

ben-mkiv, 2020-02-16:
i'm using the WebServer together with websockets and spent some time to figure out that calling addHandler on the webserver object, to attach the websockets, throws an exception if the webserver is running already.

maybe there could be added some error handling instead of crashing the whole esp

BlueAndi, 2020-02-16:
Are you adding the handler in a task with lower priority than the task where ESPAsyncWebServer is running? If yes, this might be the problem during a running webserver.
Crash dump?

ben-mkiv, 2020-02-16:
i'm adding the handler in the main loop (after initializing and starting the httpd)
no idea what priority this or the webserver runs with

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400e0e36  PS      : 0x00060930  A0      : 0x800d943c  A1      : 0x3ffbb7d0  
0x400e0e36: AsyncWebServer::addHandler(AsyncWebHandler*) at /home/esp32/esp-idf-lib/components/ESPAsyncWebServer/src/StringArray.h:68
 (inlined by) AsyncWebServer::addHandler(AsyncWebHandler*) at /home/esp32/esp-idf-lib/components/ESPAsyncWebServer/src/WebServer.cpp:74

A2      : 0x00000000  A3      : 0x3ffedb38  A4      : 0x00000003  A5      : 0x3f409afc  
A6      : 0x3ffc2d14  A7      : 0x00004000  A8      : 0x00000000  A9      : 0x3ffbb7b0  
A10     : 0x3ffe8de8  A11     : 0x400e3144  A12     : 0x00000002  A13     : 0x00000000  
0x400e3144: websocket_systemStats_Tasks::onWsEvent(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int) at /home/esp32/CLionProjects/reltrio/components/httpd-websocket/websocket_systemStats_Tasks.cpp:31

A14     : 0x00000000  A15     : 0xff000000  SAR     : 0x00000011  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000003c  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

ELF file SHA256: 9a5cee9031dc1d8fa31f9791a08d7aef2e11f1402044857d6869ce05f0b29d9c

Backtrace: 0x400e0e33:0x3ffbb7d0 0x400d9439:0x3ffbb7f0 0x400d958f:0x3ffbb940 0x400d20ca:0x3ffbb960 0x4009c2cd:0x3ffbb980
0x400e0e33: AsyncWebServer::addHandler(AsyncWebHandler*) at /home/esp32/esp-idf-lib/components/ESPAsyncWebServer/src/StringArray.h:32
 (inlined by) ?? at /home/esp32/esp-idf-lib/components/ESPAsyncWebServer/src/StringArray.h:67
 (inlined by) AsyncWebServer::addHandler(AsyncWebHandler*) at /home/esp32/esp-idf-lib/components/ESPAsyncWebServer/src/WebServer.cpp:74
```

BlueAndi, 2020-02-16:
It seems to crash by calling the LinkedListNode() constructor, called via new.
Can you link your sourcecode?

atanisoft, 2020-02-17:
Check free heap as you are likely low

ben-mkiv, 2020-02-18:
thanks @atanisoft that was the real issue, stopping the server before adding the handler just made enough memory available

closed 2020-02-18 by ben-mkiv