me-no-dev/ESPAsyncWebServer #709: AsyncWebServer over ethernet

cecco, 2020-02-17:
Hi, 
I'm trying to use this library on a ESP32 WROOM with a Wiz W5500 connected via SPI.
If i don't initialize the WiFi part i got a runtime assertion error. I can get it work over ethernet, it's configured (ping working) but no response (tcp port 80 is not open over ethernet address, only over wifi is open).

Can I use this library without WiFi but only via ethernet?

Thank You.

ivoh95, 2020-03-06:
I believe this is the same issue ive been having. 

https://github.com/espressif/arduino-esp32/issues/3580

I think we are waiting on this
https://github.com/espressif/arduino-esp32/pull/3607#issuecomment-576285941

closed 2020-05-19 by stale[bot]

k-zhai, 2020-06-12:
Any updates on this?

CircuitSetup, 2020-07-21:
@cecco or @ivoh95 did you ever figure out a solution to this?

CurtRod, 2020-12-27:
the same here... Is there any update?

ranelolesk, 2020-12-30:
Hello all! Does anyone have an update?

ArminPP, 2021-01-03:
Hello, 
I would also like to ask for a solution ...

CurtRod, 2021-01-18:
Will this give a chance to get Ethernet working with AsyncWebServer:
https://github.com/espressif/esp-idf/commit/1e905acacfa96c5c98f8f1656f27b3c4317b7c57

ZanzyTHEbar, 2024-01-18:
Any update on this?

liamcharmer, 2024-11-03:
@me-no-dev any thing that can be done for ethernet ?