me-no-dev/ESPAsyncWebServer #722: Guru Meditation Error on AsyncEventSourceClient

bbasil2012, 2020-03-05:
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400e5648  PS      : 0x00060530  A0      : 0x800e56e6  A1      : 0x3ffb1ea0  
A2      : 0x3ffdffb8  A3      : 0x3ffe33f0  A4      : 0x000004ea  A5      : 0x3ffe38da  
A6      : 0x00000a0d  A7      : 0x0a0d7d7d  A8      : 0x800e5654  A9      : 0x3ffb1e80  
A10     : 0x00000000  A11     : 0x000004ea  A12     : 0x000004ea  A13     : 0x000004ea  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
 
Backtrace: 0x400e5648:0x3ffb1ea0 0x400e56e3:0x3ffb1ec0 0x400e5703:0x3ffb1ee0 0x400e5759:0x3ffb1f00 0x400dfa31:0x3ffb1f30 0x400edf99:0x3ffb1fb0 0x4008930d:0x3ffb1fd0

PC: 0x400e5648: AsyncEventSourceClient::_runQueue() at /Users/bbasil.old/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/ESPAsyncWebServer-master/src/AsyncEventSource.cpp line 244
EXCVADDR: 0x00000008

Decoding stack results
0x400e5648: AsyncEventSourceClient::_runQueue() at /Users/bbasil.old/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/ESPAsyncWebServer-master/src/AsyncEventSource.cpp line 244
0x400e56e3: AsyncEventSourceClient::_queueMessage(AsyncEventSourceMessage*) at /Users/bbasil.old/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/ESPAsyncWebServer-master/src/AsyncEventSource.cpp line 194
0x400e5703: AsyncEventSourceClient::write(char const*, unsigned int) at /Users/bbasil.old/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/ESPAsyncWebServer-master/src/AsyncEventSource.cpp line 229
0x400e5759: AsyncEventSource::send(char const*, char const*, unsigned int, unsigned int) at /Users/bbasil.old/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/ESPAsyncWebServer-master/src/AsyncEventSource.cpp line 320
0x400dfa31: loop() at /Users/bbasil.old/Yandex.Disk.localized/sslac-esp32/sslac-esp32.ino line 552
0x400edf99: loopTask(void*) at /Users/bbasil.old/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/main.cpp line 19
0x4008930d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

Source code is like this:

AsyncWebServer server(80);
AsyncEventSource events("/events");

setup {
   server.addHandler(&events);
   server.begin();

}

loop {
if (millis()%1000==0){
    
   events.send(json_1_string.c_str(), "info1");
   events.send(json_2_string.c_str(), "info2");
   events.send(json_3_string.c_str(), "info3"); 
   events.send(json_4_string.c_str(), "info4");
   events.send(json_5_string.c_str(), "info5");
   events.send(json_6_string.c_str(), "info6"); 
   events.send(json_7_string.c_str(), "info7");
}

length of json_x_string is about 4K

closed 2020-05-18 by stale[bot]

aleuarore, 2021-06-24:
Hello,
did you find a solution for this? Thanks in advance!

zekageri, 2021-06-25:
Did you just put 7 event send func into the loop?

aleuarore, 2021-06-25:
No, my code is a little bit different, but the error is the same and I suppose the problem too.