me-no-dev/ESPAsyncWebServer #724: [BUG][Exception 28] Occurs when a file upload request is sent to a nested route

ayushsharma82, 2020-03-08:
### Issue:
Suppose we have this route for handling some multipart form data. ( simple key-value variables )
 
```
server.on("/config", HTTP_POST, [&](AsyncWebServerRequest *request){
  // Do some stuff with data here
  request->sendStatus(200);
});
```

Next, we have a nested route to handle a file:

```
server.on("/config/file", HTTP_POST, [&](AsyncWebServerRequest *request){
  // Do some stuff with data here
  request->sendStatus(200);
}, [&](AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final){
   // Handle File here
});
```

Now when we send a Multipart form-data request with a file for the `/config/file` route, AsyncWebServer will call the callback of `/config` route instead of the callback registered for `/config/file` route. 

This behaviour issues an Exception 28 Stack error because there is no 2nd lambda callback assigned for the `/config` route to handle the file.

<b>Expected Behaviour:</b> The AsyncWebServer class should look over all the assigned routes and choose a perfect match if available or if not available, then it should assign the closest match only IF there is a 2nd lambda callback assigned to handle the file.

meditant, 2020-03-12:
I have this bug when i try to upload file and redirect to /

athusrv, 2020-05-12:
I am having similar problems. Have you found a solution?

closed 2020-07-25 by stale[bot]