me-no-dev/ESPAsyncWebServer #726: Unable to access SSE from cross-origin request (CORS)

moothyknight, 2020-03-20:
Hi there, I'm trying to set up SSEs that I can access from external applications.

Here is the first error I got just trying to access the /events resource from a chrome extension:

```
Access to resource at 'http://192.168.4.1/events' from origin 'chrome-extension://cjdfilpnnecoebhgefhoaplecdboenmi' has been blocked by CORS policy: No 'Access-Control-Allow-Origin' header is present on the requested resource.
```

So I added the following to my ESP32 server code:
```
server.on("/events", HTTP_GET, [](AsyncWebServerRequest *request) {
    AsyncWebServerResponse *response = request->beginResponse(200, "text/plain", "OK");
    response->addHeader("Access-Control-Allow-Origin", "*");
    request->send(response);
});
```

which gave me the same error, so then I tried

```
server.on("/events", HTTP_GET, [](AsyncWebServerRequest *request) {
    AsyncWebServerResponse *response = request->beginResponse(200, "text/plain", "OK");
    response->addHeader("Access-Control-Allow-Origin", "*");
    response->addHeader("Access-Control-Expose-Headers", "*");
    response->addHeader("Access-Control-Allow-Credentials", "true");
    request->send(response);
  });
```

with this in the extension javascript:
`this.source = new EventSource(host+'/events', {withCredentials: true});`

which now gives me the error:
`Failed to load resource: net::ERR_CONNECTION_REFUSED`

Is it possible to accomplish this? Seems like I'm close, not quite sure how to set credentials if this is the right method, however.

moothyknight, 2020-03-20:
Okay, solved it, sorta.

I modified the library like so:

```
AsyncEventSourceResponse::AsyncEventSourceResponse(AsyncEventSource *server){
  _server = server;
  _code = 200;
  _contentType = "text/event-stream";
  _sendContentLength = false;
  addHeader("Cache-Control", "no-cache");
  addHeader("Connection","keep-alive");
  addHeader("Access-Control-Allow-Origin","*");   // <---------- Added
}
```
Is there a way to add the header in the main program instead of modifying the libs? Mostly for the sake of I don't want to have to run a very minor fork of this lib now lol.

Pablo2048, 2020-03-20:
I'm using this:
```cpp
#ifdef CORS_DEBUG
    DefaultHeaders::Instance().addHeader(F("Access-Control-Allow-Origin"), F("*"));
    DefaultHeaders::Instance().addHeader(F("Access-Control-Allow-Headers"), F("content-type"));
#endif
```

Pablo2048, 2020-03-20:
Oh, i forgot to mention: I'm using this right before `www.begin()` ...

moothyknight, 2020-03-20:
@Pablo2048 You take the cake good sir

closed 2020-03-20 by moothyknight

waymond91, 2020-12-19:
@Pablo2048  What file did you make the change in?

devfoFikiCar, 2021-12-04:
@PepeMax Even after adding that line, it does not work. Could it be due to XMLHttpRequest?

tomasz-mizak, 2022-03-07:
@FikiCarDev to solve problem with cors, single line like in Pepe answer will not sufice. Use this code below in addition.
`DefaultHeaders::Instance().addHeader("Access-Control-Allow-Headers", "content-type");`