me-no-dev/ESPAsyncWebServer #727: Separated Definition Declaration (.cpp, .h)

CatTurbo, 2020-03-23:
Hello
First a big thank to all developers, especially me-no-dev, for giving us ESPAsyncWebServer, which provides us with a great webserver for the ESP32. 
To render my code more visible (I am using PlatformeIO in VS Code by the way) I am trying my separate my code in multiple C++ files, but unfortunately ESPAsyncWebServer stops to work when I initialize it first in the header file and then provide it with a port in my cpp file. Compilation works, but I can’t access the webserver (ESP32 starts normal, gives me the IP of the server, but the page won’t load in my browser). 
My code is the following: 
In main.cpp I include my init_webserver.h and call the init_webserver function: 
`#include "init_webserver.h"
void setup()
{
	init_wlan();
	init_webserver();
}
void loop()
{}`

Then I use the init.webserver.h for the definition of my variables: 
`#ifndef INIT_WEBSERVER
#define INIT_WEBSERVER

#include <WiFi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <SPIFFS.h>
#include "struct.h"
#include "wellplate.h"

extern const char ssid[];
extern const char password[];

extern AsyncWebServer server;

void init_webserver();
void init_wlan();
String processor(const String &var);

#endif`

And finally there is my init_webserver.cpp which declares the webserver
`#include "init_webserver.h"

void init_webserver()
{
  AsyncWebServer server(80);

    if (!SPIFFS.begin(true))
    {
        Serial.println("An Error has occurred while mounting SPIFFS");
        return;
    }
    
    server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
        Serial.println("anfrage browser");
        AsyncWebServerResponse *response = request->beginResponse(SPIFFS, "/upload.htm", "text/plain");
        response->addHeader("Server", "Async Web Server");
        request->send(response);
    });

    server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
        request->send(SPIFFS, "/upload.htm", "text/html");
server.begin()
    });
`

Instead of this approach I've also tried to use pointers, 
thus I defined the pointer to AsyncWebServer in init_webserver.h with 
'extern AsyncWebServer *server'
then defined it with 
'server = new AsyncWebServer(80) 
server->on .....
server->begin()'
in the init_webserver.cpp,  however, here I got a compilation error. 

Sorry, when this is a total Noob question. 
Thank you very much in advance for help. a

BlueAndi, 2020-03-23:
This is not a problem of ESPAsyncWebserver.

You define it inside your init function, which will be destroyed after leaving the function. :-)

CatTurbo, 2020-03-24:
Oh shit.. sorry, for the stupid question. 
But only for confirmation, don't I declare it in the .h file and then define it in the .cpp. So it shouldn't be destroyed after leaving my init function, as the declaration is global, or should it?

In the meantime I've found also another method, passing the AsyncWebServer as reference to my init function (by init_webserver(AsyncWebServer &server)

BlueAndi, 2020-03-24:
Your declaration is not related to your definition, because your webserver object is located on the stack, caused by defining it inside the function. Put your webserver definition global and it will work.

Anyway its best practice today to not use global variables, but have a access function, which delivers a reference to the webserver.

You mentioned init_webserver(AsyncWebServer& server), but this is different again, because now the webserver must be defined outside your module.

CatTurbo, 2020-03-24:
Thank you very much for your Answer, BlueAndi. 
However, I am sorry, but I've lost the thread (really sorry, but I am not a professionel developper, only a guy intressted in C++ and ESP32 ;) ). 
You say that I should put my webserver definition global. That means, in my opinion, to put the 
`AsyncWebServer server(80);` part in main.cpp. So it will be global variable, like now with my `init_webserver(AsyncWebServer& server)` which is considered as bad practice. 
If I am correct, it is also not possible to put this part in a proper function and then work with access functions, as, as soon the function will be finished, the object, the server, will be destroyed, as it is out of its scope. 

Would you mind provide me with a short example what you mean with a global definition for the webserver without doing it in main.cpp and thus global?

BlueAndi, 2020-03-24:
.h
```
extern void init();
extern AsyncWebServer& getInstance();
```

Define webserver module global, see ...

.cpp
```
static AsyncWebServer myWebServer;

void init()
{
    ...
}

AsyncWebServer& getInstance()
{
    return myWebServer;
}
```

main.cpp
```
init();
AsyncWebServer& myWebServer = getInstance();
...
```

CatTurbo, 2020-03-24:
Thanks you very much, now all is clear.

closed 2020-03-24 by CatTurbo

BlueAndi, 2020-03-24:
You are welcome. :-)