me-no-dev/ESPAsyncWebServer #728: maintaining multiple websocket connections

note5, 2020-03-24:
hi,  I would like to know how to maintain multiple connections
`void onWsEvent(AsyncWebSocket *server, AsyncWebSocketClient *client, AwsEventType type, void *arg, uint8_t *data, size_t len)
{

  if (type == WS_EVT_CONNECT)
  {

    Serial.println("Websocket client connection received");
    globalClient = client;
  }
  else if (type == WS_EVT_DISCONNECT)
  {

    Serial.println("Websocket client connection finished");
    globalClient = NULL;
  }
}`
 The code above is how I am checking for connection. then the code below is how I am sending serial data roughly about every 1 second. I can get the data in one window but when I open another window the initially working window spots receivinng new data.

`if (globalClient != NULL && globalClient->status() == WS_CONNECTED)
    {
      globalClient->text(String(v));
    }`
I would really appreciate if someone could show me how to maintain all the connections

Pablo2048, 2020-03-24:
Hi,
you can use `.textAll()` method of AsyncWebSocket object to send the data to all connected clients. I'm using `.count()` method of AsyncWebSocket to asure that at least one client is connected, and then I'm using textAll(). Methods are described here https://github.com/me-no-dev/ESPAsyncWebServer#methods-for-sending-data-to-a-socket-client

note5, 2020-03-24:
This is awesome. Thank you very much.
For testing your suggestions I have modified my code as follows
`if (globalClient != NULL && globalClient->status() == WS_CONNECTED)
    {
      ws.textAll(String(v));
    }`
maybe not the best check since I am only using client

Pablo2048, 2020-03-24:
You are welcome. Better way IMHO (and the most universal one) is:
```cpp
    if (ws.count() > 0)
         ws.textAll(String(v));
```

note5, 2020-03-24:
thank you very much, it worked

closed 2020-03-24 by note5