me-no-dev/ESPAsyncWebServer #729: serveStatic on '/' responds with net::ERR_ABORTED 404 for subdirectories

mavenik, 2020-03-25:
I'm facing an issue with my static assets placed under a subdirectory and not able to load them from my web app. I searched within docs and existing issues but could not find a relevant solution. It would greatly help to point me at a solution that has already been discussed for this issue.

I'm using a DOIT ESP32 DEVKITV1 board in AP mode. I have a javascript application with the following directory structure placed under `data/web/`:

````
index.html
favicon.ico
manifest.json
static/
-css/
--main.css
-js/
--main.js
````
Server code:
 
```
server.serveStatic("/", SPIFFS, "/web/")
    .setDefaultFile("index.html")
    .setAuthentication("user", "password");
```
Observed behavior:
* All web requests to `/`, `/favicon.ico`, and `/manifest.json` work as expected. 
* `index.html` requests to load `/static/css/main.css` and `/static/js/main.js` which fail with a `net::ERR_ABORTED 404` error.
* Placing `main.css` and `main.js` up by one directory as `/data/web/static/main.css` and `/data/web/static/main.js` and requesting those resources as `/static/main.css` and `/static/main.js` respectively does not work.
* Placing `main.css` and `main.js` directly under the root directory as `/data/web/main.css` and `/data/web/main.js` works.
* Adding separate rules before `serveStatic("/")` mentioned above and as  `serveStatic("/static/css", SPIFFS, "/web/static/css/")` does not work either.

Expected behavior:
Static assets placed under subdirectories should load with an HTTP 200 OK response.

markus-fischbacher, 2020-04-14:
I'm facing nearly the same issue.
For me, serving static files from subdirectories is working, but not reliable.
Most of the times a reload of the web page leads into a 404 for different files.

xfdr0805, 2020-06-04:
the same question  me too.

markus-fischbacher, 2020-06-04:
In my case the issue #83 was the thing.
Maximum 5 requests at a time!
Browsers are sending more requests e.g. 10.
I reduced the count of files by concatenate CSS and JS files.
Now it's working like a charm.

closed 2020-08-17 by stale[bot]