me-no-dev/ESPAsyncWebServer #730: Send JSON response without using ArduinoJSON

m-menon, 2020-03-25:
Thanks for this amazing library. Just a quick question; I saw that when it comes to sending json, the examples involve ArduinoJSON. I have seen examples of `text/plain` response using `beginResponse`. So would there be anything wrong if I used `application/json` instead, for sending json? I have tested the below code and it seems to work fine but just curious because nowhere in the documentation did I find such a way of sending json (for plain text, yes).

```
// 503 => Service Unavailable
AsyncWebServerResponse *response = request->beginResponse(503, "application/json", "{\"device\":\"offline\"}");
response->addHeader("Retry-After", String(10));
request->send(response);
```

philbowles, 2020-05-21:
I think you have answered your own question already: it works. :) I use the same technique to avoid the overhead of yet another huge library.

m-menon, 2020-05-21:
thanks @philbowles. Yup, it's been working without any issues :)

closed 2020-05-21 by m-menon