me-no-dev/ESPAsyncWebServer #731: Server stops responding after some time being idle

Paytah232, 2020-03-28:
Hi I'm using an ESPAsyncWebServer on my ESP32 Sparkfun Thingy and after some time (generally vairable) the server stops responding. I added a serial print in my code using Millis() to count the time it has been running and after 106 minutes it no longer responds and I'm not sure why.

[image]

I also notice that when it no longer responds I have a random GPIO pin (16) sometimes flashes and GPIO 5 is dull-ly lit.

Any ideas for troubleshooting as ideally this server should be able to run for weeks at a time without crashing as it will have minimal traffic.

Could I implement a watchdog to reset the ESP if it doesn't get a response every 30 seconds or so?

Thanks for you help in advance!

BlueAndi, 2020-03-29:
Did you pin the ESPAsyncWebserver to a specific core? I am asking, because typically the task watchdog is enabled. But it may happened it was disabled by pinning it to a core.

Paytah232, 2020-03-29:
@BlueAndi I don't think I have pinned it to a core (as I'm not sure how to do that). I'm using it pretty much in the same way it is described by the documentation. I can send larger snippets of my code if you think it might help to diagnose the problem. If not, do you think a watchdog in the loop would stop the server timing out?

BlueAndi, 2020-03-29:
The question is whats the root cause. ESPAsyncWebserver is already observed via watchdog task. I guess the problem is something else, e.g. you mention this strange GPIO behaviour.

Is your code on github? If yes, link it here.

Paytah232, 2020-03-30:
@BlueAndi I agree I would like to know the root cause. My code is not currently on github, however I am using this as a close base (modifying headers and functions etc to suit ESP32 Async: 	https://www.eevblog.com/forum/microcontrollers/esp32-arduino-wifi-manager-file-system-(spiffs)/

There actually might be chance I am using a different release of ESPAsyncWebServer, possibly with no watchdog task? I will overwrite all files with this again just to be sure I'm using this release. 

What I notice with GPIO is after the server is unresponsive to interaction (i.e cannot open it in web browser and the loop printing function stop printing), The external led I have connected to GPIO16 stays on in a dull manor as well as GPIO 5 led (inbuilt). I have an LED on GPIO21 that does not get lit however (same as led on GPIO 16).

I restarted my computer and the server has reconnected again and functions appropriately. Will update library to this release and restart the server and see how long it function/if problems stop.

BlueAndi, 2020-03-30:
Use the head revision of ESPAsyncWebserver, which is fine.
The ESPAsyncWebserver runs in a own task, so its interesting that the loop() stops too. How do you communicate between webserver task and loop task? For me it looks like the webserver task is running, because the task watchdog doesn't reset.

You could activate the watchdog and feed it in the loop(). If you prepare a watchdog interrupt, you could print out the task stack to see where it hangs.

BlueAndi, 2020-03-30:
If you use a sw watchdog, you will automatically get a stack trace, see
https://arduino-esp8266.readthedocs.io/en/latest/faq/a02-my-esp-crashes.html#watchdog

Paytah232, 2020-05-13:
@BlueAndi Do you have an example of how to pin the server and all it's tasks to a core. I would like to isolate all server related tasks to Core 1, as I'm trying to run FastLED on Core 0, and I seem to be getting interference from somewhere.

BlueAndi, 2020-05-13:
On the ESP32 the webserver bases on AsyncTCP, therefore the necessary preprocessor defines are there: https://github.com/me-no-dev/AsyncTCP/blob/ca8ac5f919d02bea07b474531981ddbfd64de97c/src/AsyncTCP.h#L33-L37

If you are using platformio, you can define them in the platform.ini like here: https://github.com/BlueAndi/esp-rgb-led-matrix/blob/5687f6c8a13d9d0731c5b3cc0bd08305510b61f5/platformio.ini#L27-L28

Otherwise define them before you include the webserver header file.

closed 2020-07-26 by stale[bot]