me-no-dev/ESPAsyncWebServer #732: Compiling fails with build_unflags -fno-rtti to enable dynamic_cast in my code

0xFEEDC0DE64, 2020-03-28:
Hi,

I'm in the need for dynamic_cast in my code now and that is not possible, as platformio keeps adding `-fno-rtti` as a build flag by default.

This disables any runtime type informations and therefore dynamic_cast is not allowed.

philbowles, 2020-05-21:
"platformio keeps adding -fno-rtti as a build flag by default." Arduino IDE does the same and they both do it for a very good reason: the code / data pulled in by the RTTI susbsytem are simply too big and unmanageable for a tiny MCU like e.g. ESP8266.

Also, that error is useless without seeing the code that caused it...

Finally,  I am 99.9% convinced there is probably and easier / better / faster way of doing what you are trying to do with the dynamic_cast. Why not post the code / question in stackoverflow as this is a purely C++ question not related to this lib in any way.

0xFEEDC0DE64, 2020-05-25:
yeah I created a base class with lots of virtual methods named like:
* X *asX()
* Y *asY()

and they are all implemented with `return nullptr` except for class X, where it returns `this` instead. hacky workaround for dynamic_cast on my microcontroller, works like a charm, so the ticket can be closed.

closed 2020-05-25 by 0xFEEDC0DE64