me-no-dev/ESPAsyncWebServer #733: Async callback in request handler

BRAGA96, 2020-04-02:
Hi, thanks for the great library! Please tell me, can I call request->send in an asynchronous function? Simple example:
```
#include <ESPAsyncWebServer.h>

Ticker tick;

server.on("/networks", HTTP_GET, [](AsyncWebServerRequest *request) {
	tick.once_ms<AsyncWebServerRequest *>(1000, [](AsyncWebServerRequest *req) {
		req->send(200, "text/plain", "Success");
	}, request);
});
```

> * `tick.once_ms` imitates an asynchronous function

This code causes unexpected errors, does not always work, and often throws backtrace:
```
eGuru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4016ceed  PS      : 0x00060730  A0      : 0x800d4e1a  A1      : 0x3ffb77c0
A2      : 0x00000081  A3      : 0x0284bce8  A4      : 0x0284bce8  A5      : 0x3ffb77e4
A6      : 0x3ffc1880  A7      : 0x3ffb7790  A8      : 0x00000081  A9      : 0x3ffb77a0
A10     : 0x3ffd3974  A11     : 0x3f401cf9  A12     : 0x3ffd4680  A13     : 0x00000000
A14     : 0x3ffd4860  A15     : 0x3f401d7f  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
EXCVADDR: 0x000000b5  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000

Backtrace: 0x4016ceed:0x3ffb77c0 0x400d4e17:0x3ffb77e0 0x400d3b42:0x3ffb7850 0x400d3aa5:0x3ffb7890 0x400d1351:0x3ffb78b0 0x400e2538:0x3ffb78f0 0x400e259f:0x3ffb7910 0x40088b7d:0x3ffb7930
  #0  0x4016ceed:0x3ffb77c0 in AsyncClient::space() at C:\Users\BRAGA\.platformio\lib\AsyncTCP_ID1826\src/AsyncTCP.cpp:1021
  #1  0x400d4e17:0x3ffb77e0 in AsyncBasicResponse::_respond(AsyncWebServerRequest*) at C:\Users\BRAGA\.platformio\lib\ESP Async WebServer_ID306\src/WebResponses.cpp:697
  #2  0x400d3b42:0x3ffb7850 in AsyncWebServerRequest::send(AsyncWebServerResponse*) at C:\Users\BRAGA\.platformio\lib\ESP Async WebServer_ID306\src/WebRequest.cpp:798
  #3  0x400d3aa5:0x3ffb7890 in AsyncWebServerRequest::send(int, String const&, String const&) at C:\Users\BRAGA\.platformio\lib\ESP Async WebServer_ID306\src/WebRequest.cpp:798
  #4  0x400d1351:0x3ffb78b0 in createAP()::{lambda(AsyncWebServerRequest*)#1}::operator()(AsyncWebServerRequest*) const::{lambda(AsyncWebServerRequest*)#1}::_FUN(AsyncWebServerRequest*) at src/main.cpp:26
      (inlined by) _FUN at src/main.cpp:27
  #5  0x400e2538:0x3ffb78f0 in timer_process_alarm at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/esp_timer.c:443
  #6  0x400e259f:0x3ffb7910 in timer_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/esp_timer.c:443
  #7  0x40088b7d:0x3ffb7930 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

I tried several variations of asynchronous functions, not only using Ticker, for example: vector of callbacks, `ticker.attach_ms` outside of request handler, recursive `ticker.once_ms` timer. Everything leads to the same errors.

The library itself allows you to delay the creation of response asynchronously such as:
```
server.on("/networks", HTTP_GET, [](AsyncWebServerRequest *request) {
	asyncCall([request]() {
		request->send(200, "text/plain", "Success");
	});
});
```

Thanks!

hreintke, 2020-04-03:
@BRAGA96 
You cannot do this kind of async replies. 
The  AsyncWebServerRequest *request object is only available in the "on" function.
As soon as you exit the "on" function it is destructed -> in your tick function it is not possible to use it.

philbowles, 2020-05-19:
> @BRAGA96
> You cannot do this kind of async replies.
> The AsyncWebServerRequest *request object is only available in the "on" function.
> As soon as you exit the "on" function it is destructed -> in your tick function it is not possible to use it.

Not strictly true - I use it async all the time. The problem is that you are in system context when the cb fires which greatly restritcs what you can and cant do. I have my own task scheduler which runs on the main loop in "user context". It i specifically designed to serialise async events. in the on function I std::bind the request* to my handler function and stick it in the queue.

On the next loop (I don't think it actually gets invalidated untill you call rqstptr->send) the request* is still valid and my queued function runs happily, including sending back the reply on the request pointer - I have a ton of code that relies on this behaviour and works perfectly, so...

The more general point you make is a good one: I tell folk just to think of ANY callback as an ISR - Ticker won't work there either (for the same reason) - so just "set a flag and get out" or in my own case: "queue a user function and get out"

hreintke, 2020-05-21:
@philbowles 
Trying to understand how you are running the async on function.
Can you show the webserver.on(..)  with the std::bind of the request* ?

philbowles, 2020-05-23:
@hreintke 

h4 is my scheduler (global scope)
queueFunction takes a std:function<void(void)> so it can be called from anywhere with any async function, not just here

```cpp
    on("/",HTTP_GET, [this](AsyncWebServerRequest *request){
        h4.queueFunction(std::bind([this](AsyncWebServerRequest *request){ 
            H4EVENT("Root %s",request->client()->remoteIP().toString().c_str());
            _cb[wifiTag()]=stringFromInt(WiFi.getMode());
            request->send(SPIFFS,"/sta.htm",String(),false,aswsReplace);
        },request));
    });
```

And here it is working: 
[image]

PS Happy for you to msg me on FB if you require further discussion re this technique - don't want to "hijack" this thread *too* much :) just serach for "phil bowles"

tmalch, 2020-06-14:
Ok I'll hijack this thread.. this post gets mentioned and I think it needs a bit more clarification
@philbowles 
To my understanding AsyncWebServerRequest gets called from async_tcp thread whenever an event occurs on its connection (through callbacks registered on AsyncClient in ctor). So when you access AsyncWebServerRequest concurrently to the async_tcp thread from the main thread, you **potentially** run into problems.
e.g. a disconnect happens, async_tcp thread calls AsyncWebServerRequest::_onDisconnect() and the request gets deleted. In the main thread you are unaware of it, right?

Please tell me I'm wrong, as I really like your solution and it would make my life easier.

nouknouk, 2020-07-11:
Hi, I have the same kind of issue:

My ESP32 serves as an HTTP <=> InfraRed interface with my boiler.

- a client sends a request to my EspAsyncWebServer to ask for a value. 
- the `Request*` of this call is stored
- I fire an (asynchronous) request (via Infrared interface) to my boiler to get the value.
- (let's say) 1 second  later, my boiler sends me the value back. 
- It's time to fill the HTTP response with the value: in the callback called by my boiler response, I get back my `Request*` and try to do something like `request->send(200, "app/json", "{value: myValue}");`
- It crashes, I suppose because the `Request*` has been destroyed in the meanwhile.

This is a bit disappointing for me, as I thought that ESP**Async**WebServer would be capable not only of async handling of requests, but also support async filling of the responses...    

Any idea on how to handle such need ?

closed 2020-09-24 by stale[bot]