me-no-dev/ESPAsyncWebServer #734: Problem with onFileUpload(), it seems, it does not registers the callback function.

ocfu, 2020-04-03:
Hi, I can't get this simple example to work, "onUpload" is not been called when I upload a file from the /upload page. What's wrong with it?

```
AsyncWebServer server(80);

const char* ssid = "xxx";
const char* password = "xxx";

void notFound(AsyncWebServerRequest *request) {
  request->send(404, "text/plain", "Not found");
}

void onUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final) {
  Serial.println("onUpload called");
}

void setup() {

  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  if (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.printf("WiFi Failed!\n");
    return;
  }

  Serial.print("IP Address: ");
  Serial.println(WiFi.localIP());

  server.on("/", HTTP_GET, [](AsyncWebServerRequest * request) {
    request->redirect("/upload");
  });

  server.on("/upload", HTTP_GET, [](AsyncWebServerRequest * request) {
    String html = "<body><div><form method='post' action='/upload'><input type='file'><button>Send</button></form></div></body>";
    request->send(200, "text/html", html);
  });
  
  // upload a file to /upload
  server.on("/upload", HTTP_POST, [](AsyncWebServerRequest * request) {
    request->send(200);
  }, onUpload);

  server.onNotFound(notFound);
  server.onFileUpload(onUpload);

  server.begin();
}

void loop() {
}
```

Pallanez, 2020-04-04:
I think you have to set the "enctype" attribute in the HTML form element.
Also, your button seems to be doing nothing. Change to "type="submit"":

```
<form method='POST'  enctype='multipart/form-data 'action='/upload'><input type='file'><button type='submit'>Send</button></form>

```

ocfu, 2020-04-05:
Thank you Pallanez, a good point, but still, it doesn't work. Server is running on an ESP8266, Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_11_6) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/11.1.2 Safari/605.1.15 and also not working on Safari ipadOS 13.4. Using Arduino IDE 1.8.10.

here the complete code:

```
#include <Arduino.h>
#ifdef ESP32
#include <WiFi.h>
#include <AsyncTCP.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>
#endif
#include <ESPAsyncWebServer.h>

AsyncWebServer server(80);

const char* ssid = "xxx";
const char* password = "xxx";

void notFound(AsyncWebServerRequest *request) {
  request->send(404, "text/plain", "Not found");
}

void onUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final) {
  Serial.println("onUpload called");   // <-- this is never called.
}

void setup() {

  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  if (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.printf("WiFi Failed!\n");
    return;
  }

  Serial.flush();
  Serial.println();
  Serial.println();
  Serial.println("Start...");

  Serial.print("IP Address: ");
  Serial.println(WiFi.localIP());

  server.on("/", HTTP_GET, [](AsyncWebServerRequest * request) {
    request->redirect("/upload");
  });

  server.on("/upload", HTTP_GET, [](AsyncWebServerRequest * request) {
    String html = "<body><div><form method='POST' enctype='multipart/form-data' action='/upload'><input type='file' ><button type='submit'>Send</button></form></div></body>";
    request->send(200, "text/html", html);
  });
  
  // upload a file to /upload
  server.on("/upload", HTTP_POST, [](AsyncWebServerRequest * request) {
    Serial.println("/upload");    // <-- this I can see 
    request->send(200);
  }, onUpload);

  server.onNotFound(notFound);
  server.onFileUpload(onUpload);

  server.begin();
}

void loop() {
}

```

Pallanez, 2020-04-06:
Hey! I think I found the problem. 
Every HTML form you make has to have a "name" attribute  containing at least 1 char to be valid/processed.

So your valid form would look like this:

``` 
<body><div><form method='POST' enctype='multipart/form-data' action='/upload'><input type='file' name='dummy'><button type='submit'>Send</button></form></div></body>
```

HTML can get really ridiculous sometimes.

ocfu, 2020-04-07:
Pallanez, you are perfectly right! Many thanks, with this name attribute it works.

closed 2020-04-07 by ocfu