me-no-dev/ESPAsyncWebServer #735: Access variable defined outside TemplateCallback

markus-fischbacher, 2020-04-03:
Hi there, 

I want to access a variable defined outside the TemplateCallback.
It should act as a Json cache.

```
class ConfigurationJson : public DynamicJsonDocument
{
public:
    ConfigurationJson() : DynamicJsonDocument(256)
    {
    }
};

typedef std::map<String, ConfigurationJson> ConfigurationJsonMap;
typedef std::map<String, ConfigurationJson>::iterator ConfigurationJsonMapIterator;
```
```
server.on("/inputs", HTTP_GET, [](AsyncWebServerRequest *request) {
    ConfigurationJsonMap jsonCache;
    request->send(SPIFFS, "/inputs.html", String(), false, [jsonCache](const String &var) {
        if (var.equals("VERSION"))
        {
            return String(VERSION_STRING);
        }
        int dot = var.indexOf('.');
        if (dot == -1)
        {
            return var;
        }
        String context = var.substring(0, dot);
        String variable = var.substring(dot + 1);
        String filename = "/config/" + context + ".json";
        File file = SPIFFS.open(filename, "r");
        ConfigurationJson config;
        deserializeJson(config, file.readString());
        jsonCache[context] = config; // <== this line produces the error
        file.close();
        String value = config[variable].as<String>();
        Serial.println("Read " + variable + " from " + filename + " => " + value);
        return value;
    });
});
```

That is the error I get:
```
src/WebServer.cpp:53:21: error: passing 'const ConfigurationJsonMap {aka const std::map<String, ConfigurationJson>}' as 'this' argument of 'std::map<_Key, _Tp, _Compare, _Alloc>::mapped_type& std::map<_Key, _Tp, _Compare, _Alloc>::operator[](const key_type&) [with _Key = String; _Tp = ConfigurationJson; _Compare = std::less<String>; _Alloc = std::allocator<std::pair<const String, ConfigurationJson> >; std::map<_Key, _Tp, _Compare, _Alloc>::mapped_type = ConfigurationJson; std::map<_Key, _Tp, _Compare, _Alloc>::key_type = String]' discards qualifiers [-fpermissive]
    jsonCache[context] = config;
                     ^
*** [.pio/build/nodemcuv2/src/WebServer.cpp.o] Error 1
```

Is there a way to get that working?

Any help is appreciated.

Markus

philbowles, 2020-05-23:
I'm assuming you want ConfigurationJsonMap to be global? Creating an instance for every request is a *bad* idea, even if it *would* compile...

The reason it won't is that its definition requires a number of supporting functions e.g. the underlying map 'compare' function which need to be in the same scope. By declaring it inside a lambda function, the compiler expects to find the map compare function also in that scope - which of course it isn't and should not be! Hence that specific error.

Declare it globally

markus-fischbacher, 2020-05-23:
Thanks for your reply and explanation. I‘ve implemented the Singleton pattern. My main logic was implemented within a class already.

closed 2020-05-23 by markus-fischbacher