me-no-dev/ESPAsyncWebServer #742: AsyncWebServer causes SSD1306 OLED artifacts

gbaranski, 2020-04-09:
### Hardware:
Board:							WEMOS Lolin32
IDE name:						Platform.io
Upload Speed:					115200
Computer OS:                                         OS X and Windows

### Description:
I use AsyncWebServer, code for setting up server looks something like this 
```cpp
    server.on("/setAlarm", HTTP_GET, [](AsyncWebServerRequest * request) {
        if(request->hasParam("time")) {
            AsyncWebParameter* p = request->getParam("time");
            Serial.printf("Received %s with value %s from IP: \n", p->name().c_str(), p->value().c_str());
            Serial.println(request->client()->remoteIP());
            request->send(200, "text/plain", "OK");
            wifiTimeManager.saveAlarmTime(String(p->value()));
            String outputMsg = "New request!\n Time: " + String(p->value()) + "\n IP: " + request->client()->remoteIP().toString();
            lcdManager.printTextLcd("testing", 1);
        }
    });
```
As you can see on the bottom i have lcdManager.printTextLcd function called, first argument is String to print and second one is font size, function looks like this
```cpp
void ManageLcd::printTextLcd(String lcdText, int fontSize){
    clearLcd();
    display.setTextSize(fontSize);
    display.print(lcdText);
    display.display();
}
```
Here we got clearLcd(), it just clears the display
```cpp
void ManageLcd::clearLcd() {
    display.setTextColor(WHITE);
    display.setCursor(10, 0);
    display.clearDisplay();

    display.display();
}
```
Problem is when i want to print any text when request have been received, on ESP8266 i used schedule_function on async code, but i heard that on ESP32 i dont need that, and anyways schedule_function doesn't exist in ESP32 core, when i receive request i2c LCD connected to ESP prints me something like this https://imgur.com/a/hKBHXiY I also tried xTaskCreate but that gave me the same thing

OLED connected to GPIO 4 and GPIO 5, setup for lcd looks like this 

```
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, -1);

bool ManageLcd::setupLcd() {
    Wire.begin(5,4);
    if(!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) {
        Serial.println("SSD1306 allocation failed");
        return false;
    }
    delay(1000);
    display.clearDisplay();
    display.setTextColor(WHITE);
    display.setTextSize(1);
    display.display();
    delay(1000);
    return true;
}
```

darrenlock, 2020-04-13:
I had identical symptoms when I was trying to display text during a BLE Scan Callback. I also find that if I try to update the display during a AsyncWebServer request handler I occasionally get a IntegerDivideByZero error. I think it because either Wire/SSD1306 is not thread safe for async calls (I have no proof of this, however). To fix this for my BLE Scan callback I now put a message on a Queue then dequeue and process it in Loop() (Arduino).

philbowles, 2020-05-23:
" I think it because either Wire/SSD1306 is not thread safe for async calls (I have no proof of this, however)" for My 2c, that is *exactly* the problem and your solution is the one I use in *all* my ESP projects. I have my own main-loop-scheduler library for serialising this and indeed *any* async process.

spyder0069, 2020-05-24:
You can have no delays or overhead in the async routines.  As the post above just set a global flag there and then handle it in your main loop.

closed 2020-08-06 by stale[bot]