me-no-dev/ESPAsyncWebServer #745: Add URL templates match by extension

BbIKTOP, 2020-04-11:
Since regex is not working, is it possible at least to add url match by extension? Like this:

    diff WebHandlerImpl.h WebHandlerImpl.h.orig
    108,115d107
    <       if (_uri.length() && _uri.startsWith("/*.")) {
    <         String uriTemplate = String (_uri);
    <         uriTemplate = uriTemplate.substring(uriTemplate.lastIndexOf("."));
    <         if (!request->url().endsWith(uriTemplate))
    <           return false;
    <       }
    <       else

BbIKTOP, 2020-06-11:
Anyone alive here?

BlueAndi, 2020-06-11:
You could create your own handler and register it.

BbIKTOP, 2020-06-11:
@BlueAndi Sorry, don’t understand, could you explain please? How to create handler for some particular file type/extension? I tried regular expressions first, even considered they’re very heavy and consume a lot of precious resources on esp8266, but didn’t succeed since there’s a bug in the current gcc. How can i make different handlers now?
Thank you

BlueAndi, 2020-06-11:
You can add a handler via https://github.com/me-no-dev/ESPAsyncWebServer/blob/f13685ee97675be2ac9502d177d3024ebc49c1e0/src/ESPAsyncWebServer.h#L420

Inherit your handler from AsyncWebHandler, like its done for the AsyncStaticWebHandler.
This way you could create it, as you shown in your initial post.

A different way would be to register a callback for "/xxx/yyy/*" via https://github.com/me-no-dev/ESPAsyncWebServer/blob/f13685ee97675be2ac9502d177d3024ebc49c1e0/src/ESPAsyncWebServer.h#L423
In the callback you get access to the requested uri by request->url() and you can post-process it there.

The better option depends on how many different uri you will have.

BbIKTOP, 2020-06-11:
@BlueAndi but i need different handlers for different file types (i.e. file extensions)
Current url parameter does not support this, isn’t it?

BlueAndi, 2020-06-11:
Yes and no. If your handler (option 1 above) is generic enough on the interface, you implement it once, but instantiate it several times with different file extensions as parameter.

Yes, current url parameter doesn't support this.

BbIKTOP, 2020-06-11:
@BlueAndi so, why not simplify this and just add a few lines of code from my pr #751? It will provide a simple intuitive way to implement such a functionality

BlueAndi, 2020-06-11:
Not my decision, you have to wait for the maintainer. ;-)

BbIKTOP, 2020-06-11:
@BlueAndi Sure, but looks like nobody here for months...

me-no-dev, 2020-06-11:
he will show up when he has time :) preoccupied with a lot of work.

BbIKTOP, 2020-06-11:
@me-no-dev same here, mate, same here... waiting patiently, thank you!

BlueAndi, 2020-06-11:
@me-no-dev LOL

BbIKTOP, 2020-08-11:
Please, no “stale”, just add it ;)

BbIKTOP, 2020-10-11:
Please, merge it

PR #751 (2020-11-15): URL template match by extension
Fixes #745
closed 2020-11-15 by me-no-dev