me-no-dev/ESPAsyncWebServer #752: Authentication always fails on POST requests (ESP8266)

adolfintel, 2020-04-17:
This code works on GET requests, and always fails on POST request after initial authentication:
```
server.on("/aaa",HTTP_GET,[](AsyncWebServerRequest *request) {
      if(!(request->authenticate(username.c_str(),password.c_str()))) return;
      //do stuff
});
server.on("/bbb",HTTP_POST,[](AsyncWebServerRequest *request) {
      if(!(request->authenticate(username.c_str(),password.c_str()))) return;
      //do stuff
});
```

Am I doing something wrong or is this a bug? The Authentication header is sent by the browser in both requests.

philbowles, 2020-05-19:
is it this maybe #575  or this? #667 ? If you are using Basic auth, it may well be the problem

adolfintel, 2020-05-19:
I read those issues before opening this one. It could be a variant of the same problem, but in their case it was broken for GET as well.

closed 2020-08-01 by stale[bot]