me-no-dev/ESPAsyncWebServer #753: SSL TLS Server Secure problem

sergioxdev, 2020-04-18:
Hi guys, congratulations for the work you have done.

I'm trying to use the library to implement secure connections, and in particular
- server (using cer & key)
- client (adding the bearSSL function)

let's go to the problem, unfortunately I have not found examples, but looking and researching I found this post
https://github.com/me-no-dev/ESPAsyncWebServer/issues/392

I try but it doesn't work.
I'll post the code for days now that I've been trying to figure out where the problem is, but I still haven't found the solution.

`#define ASYNC_TCP_SSL_ENABLED true

#if ASYNC_TCP_SSL_ENABLED
char *cert_ = "/server.cer";
char *key_ = "/server.key";
char *password_ = NULL;
#endif

 
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <FS.h>

#if ASYNC_TCP_SSL_ENABLED
  unsigned int port_ = 443;
#else
  unsigned int port_ = 80;
#endif
AsyncWebServer server(port_);

//-----------------------------------------------
IPAddress _ap_static_ip(192,168,10,1);
IPAddress _ap_static_gw(192,168,10,1);
IPAddress _ap_static_sn(255,255,255,0);

String  _apName = "AP";
String  _apPassword = "";
String  _ssid = "Blacklight";
String  _pass = "123stella123stella";

#if ASYNC_TCP_SSL_ENABLED
int _onCertificate(void* arg, const char *filename, uint8_t **buf) {
  File file = SPIFFS.open(filename, "r");
  if (file) {
    size_t size = file.size();
    uint8_t * nbuf = (uint8_t*) malloc(size);
    if (nbuf) {
      size = file.read(nbuf, size);
      file.close();
      *buf = nbuf;
      Serial.print(F("[WEB] SSL File: "));
      Serial.print(filename);
      Serial.println(F(" OK"));
      return size;
    }
    file.close();
  }
  Serial.print(F("[WEB] SSL File: "));
  Serial.print(filename);
  Serial.println(F(" ERROR"));
  *buf = 0;
  return 0;
}
#endif

void setup() {  
  Serial.begin(115200);

  Serial.println(F("Mounting FS..."));
  if (SPIFFS.begin()) {
    Serial.println(F("Mounted file system"));
    WiFi.begin(_ssid.c_str(),_pass.c_str());
    WiFi.softAPConfig(_ap_static_ip, _ap_static_gw, _ap_static_sn);
    WiFi.enableSTA(true);
    WiFi.softAP(_apName.c_str());
    server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
      request->send(SPIFFS, "/index.html", String(), false);    
    });
    #if ASYNC_TCP_SSL_ENABLED 
      server.onSslFileRequest(_onCertificate, NULL);
      server.beginSecure(cert_, key_, password_);    
    #else
      server.begin(); // Web server start
    #endif
  }

}
void loop() {
}`

Can someone enlighten me
Thank you in advance

jsonpoindexter, 2020-04-19:
I am also getting this issue and cannot seem to find a solution online

sergioxdev, 2020-04-19:
Hi @jsonpoindexter
I just solved,
I noticed that even if: ASYNC_TCP_SSL_ENABLED 1

functions
- AsyncWebServer::onSslFileRequest
- AsyncWebServer::beginSecure
are not defined, it follows that ASYNC_TCP_SSL_ENABLED 
for the two imported libraries is always at 0

in order to resolve I had to comment
//#if ASYNC_TCP_SSL_ENABLED
//endif
in the files:
- ESPAsyncWebServer.h
- ESPAsyncWebServer.cpp
- ESPAsyncTCP.h
- ESPAsyncTCP.cpp
- SyncClient.h
- SyncClient.cpp
- tcp_axtls.h
- tcp_axtls.c

and it Works
I had to correct an error in the file: ESPAsyncTCP.cpp
line 1325: c->_recv(pcb, p->pb, 0);
in:
auto errorTracker = c->getACErrorTracker();
c->_recv(errorTracker, pcb, p->pb, 0);

It's a temporary solution but it works, look forward to @me-no-dev

jsonpoindexter, 2020-04-19:
@sergioxdev  thank you for this! I was able to compile now without errors.

I was able to find a way that does not require as many line changes. I edited lines 4-5 in `ESPAsyncTCP/src/async_config.h` to:
```
// #ifndef ASYNC_TCP_SSL_ENABLED
#define ASYNC_TCP_SSL_ENABLED 1
// #endif
```
and applied your fix in ESPAsyncTCP.cpp:
```
auto errorTracker = c->getACErrorTracker();
c->_recv(errorTracker, pcb, p->pb, 0);
```

and this worked

sergioxdev, 2020-04-19:
@jsonpoindexter

applied your fix in ESPAsyncTCP/src/async_config.h 
tested and works
:-)

closed 2020-04-29 by sergioxdev

zekageri, 2020-08-31:
Is it for the 8266?

sergioxdev, 2020-08-31:
it's about 8266

zekageri, 2020-09-02:
> it's about 8266

Thank you for the information.

yashodeepk, 2020-12-02:
Thanks @jsonpoindexter it worked for me, 
but when I  am hitting the URL in the browser and the error is

This site can’t provide a secure connection10.188.191.1 didn’t accept your login certificate, or one may not have been provided.
Try contacting the system admin.
ERR_BAD_SSL_CLIENT_AUTH_CERT

Please need a help here!!

jsonpoindexter, 2020-12-02:
Hi @yashodeepk glad to hear it worked.

It sounds possibly the issues are seeing has to do with your browser settings. This stack overflow might be relevant: https://stackoverflow.com/a/43839446

Which browser are you using?

yashodeepk, 2020-12-12:
Thanks @jsonpoindexter I will try that