me-no-dev/ESPAsyncWebServer #755: .on() not working when using in other class

jth184, 2020-04-19:
Hi,
I want to use ESPAsyncWebServer in a new project and have problems with it. I tried a first little example and it worked. Now I want to organize the web functions in a separate class, but every request is answered by the onNotFound() function. I don't have any idea, what is wrong. Can anyone please help me?

In the main.cpp:
`AsyncWebServer webserver(80);
WebHandler webHandler(webserver);

void setup() {
  Serial.begin(115200);
  WiFi.begin(ssid, pass); 
  webHandler.begin();
}`

The webHandler.h:
`#include <ESPAsyncWebServer.h>
#include "websrc.h"

class WebHandler {
    public:
    WebHandler(AsyncWebServer & webserver);
    void begin(void);

    private:
    AsyncWebServer webserver;

    void setHandlers(void);
};

And the webHandler.cpp:

`#include "webHandler.h"

WebHandler::WebHandler(AsyncWebServer & webserver) : 
webserver(webserver)
{}

void WebHandler::begin() {
    this->webserver.begin();
    setHandlers();
    this->webserver.rewrite("/", "/index.htm");
    
    Serial.println("Webhandler begin");
}

void WebHandler::setHandlers() {
    Serial.println("SetHandler");
    this->webserver.on("/index.htm",HTTP_GET, [](AsyncWebServerRequest * request) {
        Serial.println("Got request");
        AsyncWebServerResponse * response = request->beginResponse_P(200, "text/html", WEBSRC_INDEX_HTM, WEBSRC_INDEX_HTM_LEN);
        response->addHeader("Content-Encoding", "gzip");
        request->send(response);
    });
    this->webserver.on("/test",HTTP_GET, [](AsyncWebServerRequest * request) {
        AsyncWebServerResponse * response = request->beginResponse_P(200,"text/html", "Alles OK");
        request->send(response);
    });
    this->webserver.onNotFound([](AsyncWebServerRequest * request) {
        AsyncWebServerResponse * response = request->beginResponse_P(200,"text/html", "Alles ok");
        Serial.printf("Not found: %s!\r\n", request->url().c_str());
        request->send(response);
    });
    
}`

When I open the path /index.htm or /test in browser, I get the Not found page. What am I doing wrong?
Thanks a lot!

BlueAndi, 2020-04-21:
Do you copy the webserver in the WebHandler constructor by intention?

jth184, 2020-04-22:
Yes, I wanted to declare the webserver in main.cpp and then hand it over to the WebHandler. Is there something wrong with that?

BlueAndi, 2020-04-22:
It runs twice. The one you defined global and the one as member in the webhandler.

jth184, 2020-04-23:
Oh ok! I see my mistake. Now it works. Thank you very much!
I changed the line in webHandler.h from
```
AsyncWebServer webserver;
```
to
```
AsyncWebServer * _webserver;
```

and in webHandler.cpp from
```
WebHandler::WebHandler(AsyncWebServer & webserver) : 
webserver(webserver)
{}
```
to
```
WebHandler::WebHandler(AsyncWebServer & webserver) {
_webserver = &webserver;
}
```

closed 2020-04-23 by jth184