me-no-dev/ESPAsyncWebServer #756: How to assign a Class function to AsyncCallbackJsonWebHandler

darrenlock, 2020-04-21:
I am struggling to get the syntax correct to bind a Class member function to the AsyncCallbackJsonWebHandler. 

This is what I'm trying:
`AsyncCallbackJsonWebHandler* statusCRUDhandler = new AsyncCallbackJsonWebHandler("/statusCRUD", std::bind(&myClass::handleStatusCRUD, this, std::placeholders::_1));
`

The is the Class function:
```
void myClass::handleStatusCRUD(AsyncWebServerRequest* request, JsonVariant& json)
{
.....
}
```
This is the compiler error:
```
error: no matching function for call to 'AsyncCallbackJsonWebHandler::AsyncCallbackJsonWebHandler(const char [12], std::_Bind_helper<false, void (myClass::*)(AsyncWebServerRequest*, ArduinoJson6150_0000010::VariantRef&), myClass*, const std::_Placeholder<1>&>::type)
   bind(&myClass::handleStatusCRUD, this, std::placeholders::_1))
```

BlueAndi, 2020-04-21:
Using a lambda function may be easier. Didn't try std::bind by myself.

darrenlock, 2020-04-23:
Silly mistake on my part (missed the second parameter that passes the JSON):
`AsyncCallbackJsonWebHandler* statusCRUDhandler = new AsyncCallbackJsonWebHandler("/statusCRUD", std::bind(&myClass::handleStatusCRUD, this, std::placeholders::_1, std::placeholders::_2));`

closed 2020-05-23 by darrenlock