me-no-dev/ESPAsyncWebServer #758: Cannot convert 'WebRequestMethod' to 'httpd_method_t {aka http_method}' in initialization

felixgem, 2020-04-26:
I have put together two working codes and now the arduino ide does not want to compile them. The following error message is displayed:

sketch_apr24a:122:3: error: cannot convert 'WebRequestMethod' to 'httpd_method_t {aka http_method}' in initialization
exit status 1
cannot convert 'WebRequestMethod' to 'httpd_method_t {aka http_method}' in initialization

philbowles, 2020-05-21:
How can anyone eanswer this without seeing the code that cause the error?

Also, how is your (unseen) code "working codes" when it doesn't compile? Failing to compile is pretty much the best definiton of *not* working.

Show us the code with the error!

closed 2020-08-03 by stale[bot]

dnzunal, 2021-05-04:
waiting solution, i have same error