me-no-dev/ESPAsyncWebServer #763: templating for input form type checkbox fails to set checked flag

dakky, 2020-05-18:
My html snipplet:
```
<input type="checkbox" id="heartbeatEnabled" name="heartbeatEnabled" %CLOCK_HEARTBEATENABLED%>
```
the method handling the replacement:
```
String webserverProcessHtmlTemplate(const String &var)
{
    if (var == "CLOCK_HEARTBEATENABLED")
    {
        if (Config.getHeartbeat())
        {
            return String("checked");
        }
    }
}
```
so finally i expect as final result in the parsed page:
```
<input type="checkbox" id="heartbeatEnabled" name="heartbeatEnabled" checked>
```
what do I get?
```
<input type="checkbox" id="heartbeatEnabled" name="heartbeatEnabled" checked="">
``` 
In case the `Config.getHeartbeat()` is false, I get ot correct:
```
<input type="checkbox" id="heartbeatEnabled" name="heartbeatEnabled"
```

do I miss something or is this a bug?

philbowles, 2020-05-21:
There is a problem with your code: On all values *except* the heartbeat you don't return a value. This is undefined behaviour in C/C++ 

> The C++ Standard, [stmt.return], paragraph 2 [ISO/IEC 14882-2014], states the following: 

> Flowing off the end of a function is equivalent to a return with no value; this results in undefined behavior in a value-returning function.
> 
> A value-returning function must return a value from all code paths; otherwise, it will result in undefined behavior. This includes returning through less-common code paths, such as from a function-try-block, as explained in the C++ Standard, [except.handle], paragraph 15:
> 
> Flowing off the end of a function-try-block is equivalent to a return with no value; this results in undefined behavior in a value-returning function (6.6.3).
> 
>

philbowles, 2020-05-21:
Works for me exactly as you wanted:
[image]

Original HTML:

```
...
<div id="footer"><hr>
<div id="donate" class="box">
Buying me a coffee helps keep me motivated to keep H4Plugins bug-free and er ...plain "free". It's a lot of work.
If it has helped you, I'd appreciate a small sign
<form action="https://www.paypal.com/cgi-bin/webscr" method="post" target="_top">
<input type="checkbox" id="heartbeatEnabled" name="heartbeatEnabled" %CLOCK_HEARTBEATENABLED%>   
<input type="hidden" name="cmd" value="_s-xclick" />
...
```

Are you using the latest versions of everything?

dakky, 2020-05-21:
hmm you are correct. I fixed my code to return an empty string if the feature is disabled => it works fine. Thanks a lot!

closed 2020-05-21 by dakky