me-no-dev/ESPAsyncWebServer #769: Can't "begin" a server after its "end"

MassiPi, 2020-05-24:
Hello,
i'm trying to achieve this:

- when the esp8266 connects to wifi, it setups a web server with the pages needed for the common operation (so, the "proper" begin())
- when the esp8266 can't connect to wifi - so BEFORE the above - it starts an AP and serves (first begin(), with custom pages) a page where enter the configuration data.
- when new data is received, the webserver is ended (end()) and it goes back to the first point, where the "proper" server should be started.

The problem is that when i end the server, i can't wake it up again. 

Well, i ended up using two different Asyncwebserver vars, given they never are up together, but it seems non a clean solution.
Why do i need an end() method if i can't begin it again?
thanks

PS: is there a way to see if a response is over or is still being sent? (to check when it's safe to end a server.. i ended up in this case with a delay..)

MassiPi, 2020-05-27:
well, i ended up with a single server, filtering on the interface with the .setfilter() method.
I still wonder if is there a way to know if the server has some responses ongoing..

IoTThinks, 2020-06-05:
I have the same issue.
Have you found a workaround?

IoTThinks, 2020-06-05:
Or we can disable/enable handlers instead of begin/end server?
// remove all rewrites, handlers and onNotFound/onFileUpload/onRequestBody callbacks
server.reset();

MassiPi, 2020-06-08:
> Have you found a workaround?

Well, i ended up with a single instance, but with different pages served on differente interfaces (AP or STA) through the filters the library has.
This obviously can only work for my specific case, where i wanted to separate the two interfaces..
in general no, i can't make it work.

IoTThinks, 2020-06-10:
I reset the WiFi before begin() again.
Then it works.

I believe there is "port is busy" issue.

closed 2020-08-23 by stale[bot]