me-no-dev/ESPAsyncWebServer #772: E (10080) task_wdt: Task watchdog got triggered.  - async_tcp (CPU 1)

mrWheel, 2020-05-31:
###Hardware:
Board: ESP32 Dev Module
Core Installation version: 1.0.4
IDE name: Arduino IDE v1.8.10
Flash Frequency: 80Mhz
PSRAM enabled: no
Upload Speed: 115200
Computer OS: Mac OSX

###Description:
Stress testing a simple webserver.
From multiple Webbrowser stressTest.html reloads every 3 seconds loading index.css and stressTest.js. I sometimes get errors stating the loaded files have a length of 0 bytes. If thats the case it seems the WiFi connection is lost but the loop() function keeps running (showing a blinking led);

Looks like [this](https://github.com/espressif/arduino-esp32/issues/3000) problem

###Debug Messages:
<pre>
[13:29:18][ 204868|113792] restAPI     (  29): from[192.168.12.20] URI[/api/v0/devinfo] method[GET] 
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[13:29:19][ 212212|113792] operator()  ( 109): in 'onNotFound()'!! [/favicon.ico] => 
[13:29:19][ 212212|113792] operator()  ( 126): next: handleFile(/favicon.ico)
[13:29:19][ 204920|113792] restAPI     (  29): from[192.168.12.153] URI[/api/v0/devinfo] method[GET] 
[E][vfs_api.cpp:238] VFSFileImpl(): fopen(/spiffs/stressTest.js) failed
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[13:29:20][ 210428|113792] restAPI     (  29): from[192.168.12.20] URI[/api/v0/devinfo] method[GET] 
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[13:29:22][ 203660|113792] restAPI     (  29): from[192.168.12.153] URI[/api/v0/devinfo] method[GET] 
[13:29:23][ 194244|113792] restAPI     (  29): from[192.168.12.20] URI[/api/v0/devinfo] method[GET] 
[E][vfs_api.cpp:238] VFSFileImpl(): fopen(/spiffs/stressTest.js) failed
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[13:46:52][ 207628|113792] operator()  ( 109): in 'onNotFound()'!! [/favicon.ico] => 
[13:46:52][ 207628|113792] operator()  ( 126): next: handleFile(/favicon.ico)
[E][vfs_api.cpp:238] VFSFileImpl(): fopen(/spiffs/FSexplorer.html) failed
[E][vfs_api.cpp:238] VFSFileImpl(): fopen(/spiffs/favicon.ico) failed
[E][vfs_api.cpp:238] VFSFileImpl(): fopen(/spiffs/favicon.ico) failed
[E][vfs_api.cpp:238] VFSFileImpl(): fopen(/spiffs/favicon.ico) failed
[E][vfs_api.cpp:238] VFSFileImpl(): fopen(/spiffs/favicon.ico) failed
[E][vfs_api.cpp:238] VFSFileImpl(): fopen(/spiffs/favicon.ico) failed
[E][vfs_api.cpp:238] VFSFileImpl(): fopen(/spiffs/stressTest.html) failed
[E][vfs_api.cpp:238] VFSFileImpl(): fopen(/spiffs/FSexplorer.png) failed
[E][vfs_api.cpp:238] VFSFileImpl(): fopen(/spiffs/settings.png) failed
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[13:47:02][ 210980|113792] restAPI     (  29): from[192.168.12.20] URI[/api/v0/devinfo] method[GET] 
[13:47:05][ 212812|113792] restAPI     (  29): from[192.168.12.20] URI[/api/v0/devinfo] method[GET] 
</pre>
With `**SPIFFS.begin(false,"/spiffs",30)**`
<pre>
[13:16:06][ 209048|113792] restAPI     (  29): from[192.168.12.20] URI[/api/v0/devinfo] method[GET] 
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
E (10080) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10080) task_wdt:  - async_tcp (CPU 1)
E (10080) task_wdt: Tasks currently running:
E (10080) task_wdt: CPU 0: IDLE0
E (10080) task_wdt: CPU 1: IDLE1
E (10080) task_wdt: Aborting.
abort() was called at PC 0x400edd77 on core 0

Backtrace: 0x4008c5c0:0x3ffbe170 0x4008c7f1:0x3ffbe190 0x400edd77:0x3ffbe1b0 0x40084771:0x3ffbe1d0 0x40168b97:0x3ffbc250 0x400ef12e:0x3ffbc270 0x4008a361:0x3ffbc290 0x40088b7d:0x3ffbc2b0

================decode====================
Decoding stack results
0x4008c5c0: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008c7f1: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x400edd77: task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c line 174
0x40168b97: esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c line 492
0x400ef12e: esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c line 63
0x4008a361: prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 3382
0x40088b7d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
==========================================

Rebooting...
</pre>
What is there to do to make this robust?

BlueAndi, 2020-06-01:
Just a note, calling delay() in any webserver callback blocks the AsyncTcp thread, which is bad.

mrWheel, 2020-06-01:
> Just a note, calling delay() in any webserver callback blocks the AsyncTcp thread, which is bad.

The delay()’s are there as an attempt to see if it changes anything. Without the delay’s the outcome is the same.

> SPIFFS is very slow, which can cause a task watchdog, depended on file size. To increase robustness, you could queue requests and handle them in the main thread.

How can I do that?

BlueAndi, 2020-06-01:
Using a FreeRTOS queue (see xQueueCreate()) and e.g. queue a lambda function in and bind the request object to it. See a example here: https://github.com/me-no-dev/ESPAsyncWebServer/issues/733#issuecomment-633015579

The example uses a own queue functionality.

closed 2020-08-15 by stale[bot]