me-no-dev/ESPAsyncWebServer #773: #ifndef WEBSERVER_H

UpperAlex, 2020-05-31:
Hey,

I got the error of redeclaration of HTTP_GET etc while using another library. Your ifndef works for the EPS32 implementation of the WebServer. But the ESP8266WEBSERVER_H is not respected.
Would be a helpful and fast fix i guess. (ESPAsyncWebServer.h line 54) 
If there is a better way, would be greate to hear it. 

Greetings.

My work-around: 
`
#ifdef ESP32
  #ifdef WEBSERVER_H
    #define HTTP_ENUM
  #endif
#elif defined(ESP8266)
  #ifdef ESP8266WEBSERVER_H
    #define HTTP_ENUM
  #endif
#endif
#ifndef HTTP_ENUM
typedef enum {
  HTTP_GET     = 0b00000001,
  HTTP_POST    = 0b00000010,
  HTTP_DELETE  = 0b00000100,
  HTTP_PUT     = 0b00001000,
  HTTP_PATCH   = 0b00010000,
  HTTP_HEAD    = 0b00100000,
  HTTP_OPTIONS = 0b01000000,
  HTTP_ANY     = 0b01111111,
} WebRequestMethod;
#endif
`

dontsovcmc, 2020-05-31:
dublicate: https://github.com/me-no-dev/ESPAsyncWebServer/issues/760

pranjal-joshi, 2020-07-06:
Resolved: https://github.com/me-no-dev/ESPAsyncWebServer/issues/760#issuecomment-654108029

closed 2020-07-06 by UpperAlex