me-no-dev/ESPAsyncWebServer #776: [How to] Use Arduino variables with PROGMEM R"rawliteral(

pranjal-joshi, 2020-06-01:
Hello,

I have a big HTML page which is declared as below:
```
const char saved_ok_html[] PROGMEM = R"rawliteral( 
// some big HTML code here // 
)rawliteral";
```
Here, I want put values of some of the variables declared globally into the webpage while sending.
Something like this
```
R"rawliteral( // Some HTML // )rawliteral + myTemperatureVariable + R"rawliteral( // Some HTML // )rawliteral"
```

**How to do this so that variable data will be added to the webpage in real time?
Keep in mind that I want to keep the HTML page on PROGMEM!**

pranjal-joshi, 2020-06-02:
@me-no-dev Gentle reminder, I am eagerly waiting for solution.

atanisoft, 2020-06-02:
@pranjal-joshi Guess you didn't look in the readme: https://github.com/me-no-dev/ESPAsyncWebServer#send-large-webpage-from-progmem-containing-templates

pranjal-joshi, 2020-06-03:
@atanisoft Thanks for pointing me towards the solution! I got this working with a processor.
Closing this now as the issue is resolved!

closed 2020-06-03 by pranjal-joshi

floriangerbaud, 2022-05-29:
@pranjal-joshi then how did you find the issue ?

moiHocArduino, 2022-08-13:
hi **Pranjal-joshi**, can you please tell me how to do it. thank you very much.
i have read through "readme" but haven't found a way to solve this problem.

kalex3000, 2022-11-01:
use  %Text% inside  R"rawliteral( 

and outside 
a function
String processor(const String& var){
  if(var == "Text"){  
    String includeText=""; 
   // code to do it 
    return includeText;
  }
return String();
}