me-no-dev/ESPAsyncWebServer #777: Stop and resume handle server

papirrin999, 2020-06-01:
this library needs pause and resume the handling of service.

markus-fischbacher, 2020-06-04:
I'm using begin() and end() functions for that purpose.
Is working well for me.

IoTThinks, 2020-06-05:
When I used end(), the web server can not be on using begin() anymore.
@markus-fischbacher Did you check the end() then begin() to see if the web server is up again?

markus-fischbacher, 2020-06-05:
@IoTThinks for me it‘s definitely working.

IoTThinks, 2020-06-05:
I use server.reset() to stop servicing requests and add handler again when want to start servicing request again.
Work for me.
// remove all rewrites, handlers and onNotFound/onFileUpload/onRequestBody callbacks
server.reset();

papirrin999, 2020-06-06:
I think he was right, because some people need to stop and resume it, and they resort to other functions that are not designed for that purpose, including you.

IoTThinks, 2020-06-06:
As long as it fits my need hehe.
I suspect it is port-in-use issue.

markus-fischbacher, 2020-06-20:
After some testing I could recognise, that begin() is not working reliable. Sometimes the Webserver is not running. After additional end()/begin() call it starts working again. I need to find a solution for that. I think it occurs, when a Client is connected to the server.

IoTThinks, 2020-06-23:
Welcome to the same boat. :D
At least, the issue can be detected.

markus-fischbacher, 2020-06-23:
@IoTThinks how do you manage to detect the issue? Are you trying to connect to an endpoint?

IoTThinks, 2020-06-23:
@markus-fischbacher I have an esp32 as battery node.
I use this library to turn on the web server for configuration (sensors...)
Then after done, I push the button to turn this web server down to save battery.
Then I push the button again to begin the web server to configure again. Then tada, the web server can not be on again.

The web server can only be on again if I reset the Wifi before turning on the web server.
Hence, I suspect that the system still hold on the port 80 when we do webserver.end().
And when we turn on the web server, it can not get the port 80.

markus-fischbacher, 2020-06-23:
Ok, the idea with the blocked port sounds possible. On Windows the port can held some time before it get‘s released. Probably it‘s the same on ESP OS. Unfortunatley the library does not return the error code of tcp_bind() in AsyncServer. It just returns without the possibility to get the last error. That could be ERR_ISCONN or ERR_USE. The latter one would be your assumtion.

markus-fischbacher, 2020-06-23:
I exposed the AsyncServer to the AsyncWebServer class and checking the status after calling end(). I wait till the status is returning CLOSED. After that, the port should be available again. I can't reproduce the error now. Have to wait till it occurs again.

```
	void webServerEnd()
	{
		TRACE_ME
		this->server->end();
#ifdef SERIAL_DEBUG_WEB
		Serial.print(F("Closing WebServer"));
#endif
		uint8_t retries = 50;
		while (this->server->server()->status() != wl_tcp_state::CLOSED && --retries >= 0)
		{
			delay(100);
#ifdef SERIAL_DEBUG_WEB
			Serial.print(".");
#endif
		};
#ifdef SERIAL_DEBUG_WEB
		Serial.println(" => closed");
#endif
		this->server = nullptr;
	}
```

BlueAndi, 2020-06-23:
I have a similar problem. After the wifi connection is lost, I stop the webserver (end()) and after successful reconnect, I start it again (begin()). The task which handles the event queue for all LwIP events starts and unexpectedly stops sometimes (checked by printing a '.' in the task for poll event).

On my side its not a problem regarding the pcb (port), which may be in time-wait state (https://github.com/esp8266/Arduino/tree/master/doc/faq#how-to-clear-tcp-pcbs-in-time-wait-state-), as you mentioned with tcp_bind(). Typically tcp_bind() will return -8 (https://github.com/espressif/esp-lwip/blob/5c181728c894d8c051070b1cf21c55a6222fc64b/src/include/lwip/err.h#L70-L71).

markus-fischbacher, 2020-06-23:
Setting the socket option CONFIG_LWIP_SO_REUSE could be a way to go.
On ESP32 platform it's a sdk configuration. If using Arduino on ESP8266, where can CONFIG_LWIP_SO_REUSE be configured/defined?

IoTThinks, 2020-06-24:
@markus-fischbacher On esp32, what setting could we use? 
In Google, I see some of them use "#define CONFIG_LWIP_SO_REUSE 1"

markus-fischbacher, 2020-06-24:
Yes, should be possible.

BlueAndi, 2020-06-24:
Arduino based with platformio on esp32, its not possible, because the LwIP stack comes as binary.

closed 2020-09-06 by stale[bot]

array81, 2021-02-06:
Is there a solution about this problem?