me-no-dev/ESPAsyncWebServer #779: OTA with AsyncWebServer

nvecchio1, 2020-06-02:
Hi,
is it possibile update my firmware on esp32 on wifi with this library?
If yes, how can i?
Let me know please.

lorol, 2020-06-02:
Yes you can. 
See the the official OTA example that comes with Arduino IDE for ESP8266 and ESP32:
File->Examples->ArduinoOTA->BasicOTA

nvecchio1, 2020-06-03:
Sorry, i want to update my firmware over wifi with http request.
Is it possible with this library?
I saw that with a normal webServer it si possible, but with asyncWebServer?

dushyantahuja, 2020-06-03:
yes - you can use this https://github.com/ayushsharma82/AsyncElegantOTA

markus-fischbacher, 2020-06-05:
There is an example available here: https://github.com/me-no-dev/ESPAsyncWebServer#setting-up-the-server

closed 2020-08-18 by stale[bot]

dpetican, 2022-03-13:
> There is an example available here: https://github.com/me-no-dev/ESPAsyncWebServer#setting-up-the-server

This example only works on the ESP8266.  I've been using it for years on an RSP8266.  The ESP class is not defined on the latest ESP32 Arduino framework and/or IDF.  Which is not helpful.  In fact although ESPAsyncWebserver has been very good for me I've found upgrading my project to the ESP32 has been difficult.  Especially if using the SdFat library to make an SD webserver.  AsyncElegantOTA is a possible solution for me, but the creakiness of this project has me concerned going forward.