me-no-dev/ESPAsyncWebServer #780: SPIFFS being deprecated

dushyantahuja, 2020-06-03:
Hi, I get the following warning while compiling:
```
warning: 'SPIFFS' is deprecated (declared at **** ): SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems. [-Wdeprecated-declarations]
   AsyncWebServerResponse *response = request->beginResponse(SPIFFS, "/www/timelapse.html", "text/html");
```
Is there any plan to move from SPIFFS to LittleFS?

BlueAndi, 2020-06-04:
https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html#:~:text=SPIFFS%20is%20currently%20deprecated%20and,times%20faster%20for%20most%20operations.

It has nothing to do with the ESPAsyncWebServer.

dushyantahuja, 2020-06-04:
It does - ESPAsyncWebServer has methods that take SPIFFS, see examples here:
https://github.com/me-no-dev/ESPAsyncWebServer#respond-with-content-coming-from-a-file

also there is the SPIFFEditor.h

BlueAndi, 2020-06-04:
SPIFFS is only a example there, you could use other FS as well.
See https://github.com/me-no-dev/ESPAsyncWebServer/blob/f13685ee97675be2ac9502d177d3024ebc49c1e0/src/WebRequest.cpp#L775

SPIFFSEditor uses fs::FS so it should work with other FS too. The name may be missleading from my point of view.

Anyway I hope platformio will support LittleFS in the future too, so it could easy be used with ESPAsyncWebServer. :-)

lorol, 2020-06-04:
It works fine with LittleFS on ESP8266, It needs only few tweaks of SPIFFSEditor to handle the fact that LittleFS has folders.

BlueAndi, 2020-06-04:
@lorol Do you use platformio?

lorol, 2020-06-04:
@BlueAndi for LittleFS validation I just used the Arduino IDE

lorol, 2020-06-05:
Here is the LittleFS example:
https://github.com/lorol/ESPAsyncWebServer/tree/master/examples/ESP_AsyncFSBrowser

dushyantahuja, 2020-06-05:
Thanks - I like the:
```
#ifdef USE_LittleFS
	#include <FS.h>
	#define SPIFFS LittleFS
	#include <LittleFS.h> 
 #endif
```
This should get existing code to run without making any further changes.

closed 2020-06-05 by dushyantahuja

lorol, 2020-06-05:
Yes, it will run. However SPIFFS has no real folders while LittleFS has and you need to resolve slashes issues  and iterate files in the folders where required.
Indeed a great work is done with this wrapper, hope we will see soon a similar clear, nice and explained with examples for ESP32 that allows reusing of code for SPIFFS :
https://github.com/atrimper/meterrific_firmware/blob/master/WiFi/Arduino/hardware/esp8266com/esp8266/libraries/LittleFS/src/LittleFS.h

Of course an uploading tool, too.

BlueAndi, 2020-06-05:
Regarding upload, there is already a script available, see discussion here:
https://community.platformio.org/t/how-to-define-littlefs-partition-build-image-and-flash-it-on-esp32/11333

gnalbandian, 2020-09-23:
Does anybody has an tweaked version of SPIFFSEditor that works with LittleFS and is willing to share it?
@lorol maybe?

thanks..

lorol, 2020-09-23:
@gnalbandian 
Yes, my fork of ESPAsyncWebServer supports it.
Please use the previous commits, before I merged a pull request from other member - it needs some fixes.

gnalbandian, 2020-09-23:
Thanks @lorol. I really appreciate your help.
Will you submit this commits to the original repo?

lorol, 2020-09-23:
The official repo is frozen since last year. I don’t think it will have updates anymore.

lorol, 2020-09-24:
As promised a PIO example demonstrating LittleFS on esp32:  https://github.com/lorol/LITTLEFS/tree/master/examples/LITTLEFS_PlatformIO

Duckle29, 2021-03-22:
@lorol Will you consider [publishing](https://docs.platformio.org/en/latest/librarymanager/creating.html#publishing) your fork on the PIO registry so that it can be easily used? :)

yuyubades, 2021-05-06:
- Edit ESPAsyncWebServer.h and SPIFFSEditor.cpp change #include <FS.h> to #include <LittleFS.h>
- Edit SPIFFSEditor.h find line 16:
 change const fs::FS& fs=SPIFFS to const fs::FS& fs=LittleFS

Anyeos, 2022-11-07:
For anyone like me searching for LittleFS support on SPIFFSEditor for ESP8266 on Arduino: The FS parameter is at the end of the function unlike the one for ESP32. LoL I just lost a lot of time to look at the last parameter and note that there is. :-S LoL :-P

MacDada, 2023-05-13:
> The official repo is frozen since last year.

@lorol What? It isn't mentioned anywhere in the `README.md`…