me-no-dev/ESPAsyncWebServer #781: send to client

jimhoranjr, 2020-06-04:


markus-fischbacher, 2020-06-05:
What do you want to do exactly?

There are at least two possibilities:
1. Use Ajax to request further data initiated by the client
2. Implement WebSocket to push data to the client

jimhoranjr, 2020-06-05:
I saw a library to use a webpage as a serial monitor. My knowledge of  HTML
is fairly rudimentary and I thought I could learn some more trying to replicate it. 
Your options are  what I had run across so far. I just wanted to make sure I wasn't
missing something easier. 
    I've seen sketches using Ajax to automatically update a webpage using both 
the webserver.h library and your library. It appears to me that the differences 
in the sketches are that with the webserver.h library  some code in the loop is 
necessary  to keep rechecking the client. With the ESPAsyncWebServer.h
library this isn't necessary. Am understanding this correctly?

Thanks for your work.

closed 2020-08-18 by stale[bot]