me-no-dev/ESPAsyncWebServer #792: LittleFS on ESP32, feedback needed

lorol, 2020-06-17:
Hi I made a LittleFS for ESP32, too. see the example my library and tool plus the core conversation.
Please test and comment. (performance, speed, stability, code size, slashes, directories and root / as well as other compatibility concerns)

Example (modified from the master ESP_AsyncFSBrowser)
Note I did tweaks of Editor as well to handle folders and slashes
https://github.com/lorol/ESPAsyncWebServer/tree/master/examples/ESP_AsyncFSBrowser

Library:
https://github.com/lorol/LITTLEFS

Tool:
https://github.com/lorol/arduino-esp32littlefs-plugin

Core discussion:
https://github.com/espressif/arduino-esp32/issues/3765

Simple usage:
#include <FS.h>
#define SPIFFS LITTLEFS
#include <LITTLEFS.h>

BlueAndi, 2020-06-17:
Could you add a library.json too for platformio?

lorol, 2020-06-17:
@BlueAndi  - Don, just added an improvised one. I have no idea if it makes sense. Please check and let me know.

BlueAndi, 2020-06-17:
A first try showed:

The "esp_littlefs.h" includes the "sys/termios.h" which contains baudrate preprocessor defines that clashes with the binary number preprocessor defines of "binary.h" (included in Arduino.h).

I see you need at least IDF v3.3 based stuff, because e.g. the esp_vfs_t doesn't match at my side currently. It looks like I still get IDF v3.2 via platformio. :-(

lorol, 2020-06-17:
The IDF Arduino is compiled on is this v.3.3: git checkout
46b12a560a29fa6ade07800a4abe12a026183988
The original wrapper https://github.com/joltwallet/esp_littlefs was
modified not long ago so it likely is tested on even newer IDF.
If PIO uses older IDF, you need to resolve it yourself.
My target was to work on Arduino IDE.
For me, PIO looks like a F1 car we try to drive with your home car licence
:)

lorol, 2020-06-17:
@BlueAndi - I got what you say ... 
Well I used the git version of the core,  I need to note it somewhere

BlueAndi, 2020-06-17:
With PIO I get more possibilities, as you mention as F1 car and I use them. ;-)
I will try it again, if I get IDF 3.3. Anyway thanks for your effort.

lorol, 2020-06-21:
@BlueAndi  
Try now, but first in **esp_littlefs.c** un-comment the line:  
 #define CONFIG_LITTLEFS_FOR_IDF_3_2
At least it works now with arduino release core 1.0.4.
Let me know if PIO likes it this way :)

BlueAndi, 2020-06-22:
Ok, thanks. I will try and report back. Because I am very busy, I may need some more days.

lorol, 2020-06-22:
I updated LITTLEFS in GitHub to be with #define CONFIG_LITTLEFS_FOR_IDF_3_2 by default, which allows build on several IDF versions, including the release 1.0.4 and POI on Arduino platform - it works.

zekageri, 2020-06-23:
It will be really good if i would be able to upload little fs just like spiffs on platformio vscode ( upload file system image ).

I'am really looking for that littlefs because the spiffs can't handle the 16mb partition on my esp32. I will try it out definietly!

lorol, 2020-06-23:
@zekageri 
Just try for fun:
- close vscode
- In your PIO installation, find a file, named "mkspiffs_espressif32_arduino"  [.exe]  :)
- Go to: https://github.com/earlephilhower/mklittlefs
- Download the mklittlefs[.exe] executable for your OS from releases there
- rename your original mkspiffs_espressif32_arduino[.exe] to sometting else like mkspiffs_espressif32_arduino_bak[.exe]
- put the mklittlefs[.exe] in same folder where mkspiffs_espressif32_arduino_bak[.exe] is
- rename mklittlefs[.exe] to mkspiffs_espressif32_arduino[.exe] 
- Re-open your vscode project
- have data folder prepared
- Upload File System Image as usual 
- Enjoy
- Don't forget to restore against SPIFFS 
- Appreciate if you find the right people to implement it properly in PIO (different command, choise, prompt, confing in the ini .. etc)

zekageri, 2020-06-24:
i'am trying it right now. I have an error currently wich is :

```
src/Includes.h:32:16: error: conflicting declaration 'fs::SPIFFSFS LITTLEFS'
 #define SPIFFS LITTLEFS
```

It looks like LITTLEFS defined somewhere already.

lorol, 2020-06-24:
use SPIFFS.begin() - with default parameters. not  .begin(something)

zekageri, 2020-06-24:
> use SPIFFS.begin() - with default parameters. not .begin(something)

I get this:
```
if(!LITTLEFS.begin(FORMAT_LITTLEFS_IF_FAILED)){
        Serial.println("LITTLEFS Mount Failed");
        return;
    }
```

from LITTLEfs examples.

lorol, 2020-06-24:
Check the scope of #define  where you do a redefinition of SPIFFS to LITTLEFS . i.e. put it to the sketch, not in .h file ... etc
Also SPIFFS.h should not be included anywhere in your project, only LITTLEFS.h

zekageri, 2020-06-24:
Yeah i know. Some of the libs using SPIFFS.h. Such as JPEGDecoder as i mentioned. But it was just a problem for about a minute, while i renamed every SPIFFS occourences to LITTLEFS. Now, it seems to mee that everything is working. HTTP file upload, page loads, read write, tft images from fs etc.

zekageri, 2020-06-26:
Okay thats weird.

**The thing is that if i open a web page on the esp, and not exiting from it. ( this webpage is without websockets ) and open an other webpage from an other device, the esp crashes with this exception code:**

```
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
E (48880) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (48880) task_wdt:  - async_tcp (CPU 0/1)
E (48880) task_wdt: Tasks currently running:
E (48880) task_wdt: CPU 0: IDLE0
E (48880) task_wdt: CPU 1: loopTask
E (48880) task_wdt: Aborting.
abort() was called at PC 0x400fcba4 on core 0

0x4008cd30: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008cf61: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x400fcba4: task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c line 174
0x40176ac7: esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c line 492
0x400fe102: esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c line 63
0x4008abdd: prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 3382
0x400893e9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

I have no files that does not exist, everything is there, like when i used SPIFFS.

### This is where the program crashes:

```
server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
    AsyncWebServerResponse* response = request->beginResponse(LITTLEFS, "/Web_App.html", "text/html");
    response->addHeader("Content-Encoding", "gzip");
    request->send(response);
});
```

**This web is asking for three files when it is loaded. But it does not reach the point where it does actually.**

zekageri, 2020-06-26:
Oh. It does not happen on windows. And if i open it on android, from the browser, it is loading fine. But if i put the icon into my home screen with chrome, and i open it from there, it is crashing. Wtf

me-no-dev, 2020-06-26:
interesting... seems like Android creates some extra request or something. Did you add onNotFound callback to see what other URLs might be requested? Maybe add a print line so you know when the index is requested?

zekageri, 2020-06-26:
### I have this:

```
server.onNotFound([](AsyncWebServerRequest *request){
  AsyncWebServerResponse* response = request->beginResponse(LITTLEFS, "/NotFound.html", "text/html");
    response->addHeader("Content-Encoding", "gzip");
    request->send(response);
});
```

It is sending a page. All of my files are gzipped. Without the .gz extension so IOS can handle it properly.
But i will investigate it further.

zekageri, 2020-06-26:
### I have on the pio ini:

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino

upload_protocol = espota
upload_port = 192.168.0.33
;upload_port = COM3
monitor_speed = 115200
build_type = debug
monitor_filters = esp32_exception_decoder

board_build.flash_size = 16MB (128mb)
board_build.flash_mode = dio
board_build.partitions = large_spiffs_16MB.csv
build_flags = -DCORE_DEBUG_LEVEL=5
```
### In the onNotFound callback:

```
int notfound = 0;

server.onNotFound([](AsyncWebServerRequest *request){
  notfound++;
  Serial.println(notfound);
  AsyncWebServerResponse* response = request->beginResponse(LITTLEFS, "/NotFound.html", "text/html");
  response->addHeader("Content-Encoding", "gzip");
  request->send(response);
});
```

**Serial does not print anything. Just this exception code when crashing:**

```
E (43484) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (43484) task_wdt:  - async_tcp (CPU 0/1)  
E (43484) task_wdt: Tasks currently running:
E (43484) task_wdt: CPU 0: IDLE0
E (43484) task_wdt: CPU 1: loopTask
E (43484) task_wdt: Aborting.
abort() was called at PC 0x400fd28c on core 0

Backtrace: 0x4008cd30:0x3ffbe170 0x4008cf61:0x3ffbe190 0x400fd28c:0x3ffbe1b0 0x40084ec9:0x3ffbe1d0 0x401771af:0x3ffbc1b0 0x400fe7ea:0x3ffbc1d0 0x4008abdd:0x3ffbc1f0 0x400893e9:0x3ffbc210
  #0  0x4008cd30:0x3ffbe170 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x4008cf61:0x3ffbe190 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #2  0x400fd28c:0x3ffbe1b0 in task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c:252
  #3  0x40084ec9:0x3ffbe1d0 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
  #4  0x401771af:0x3ffbc1b0 in esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c:492
  #5  0x400fe7ea:0x3ffbc1d0 in esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c:108
  #6  0x4008abdd:0x3ffbc1f0 in prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
  #7  0x400893e9:0x3ffbc210 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

zekageri, 2020-06-26:
I have tried the tweaked version of the async lib by lorol, but that did not helped.

zekageri, 2020-06-26:
### The code not even reach this point, where the page should be served:

```
server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
  notfound++;
  Serial.print("NOT FOUND CRASH HERE: ");
  Serial.println(notfound);
    AsyncWebServerResponse* response = request->beginResponse(LITTLEFS, "/Web_App.html", "text/html");
    response->addHeader("Content-Encoding", "gzip");
    request->send(response);
});
```

zekageri, 2020-06-26:
**Oh god. Sorry i was dumb AF.
I'am using modbus RTU on the Serial and when i commented out the modbus init, i don't Serial.begin(); therefore i don't get any print. The code reach the root point, but when serving, it is crashing.** 

### Zero not found callback happens at that time.

lorol, 2020-06-28:
@zekageri 
Do the things work as expected now on your side?

zekageri, 2020-06-29:
> @zekageri
> Do the things work as expected now on your side?

Nop. The ESP still crashes if i open the webpage from my phone.
Interestingly, now i tried with some other phones and PCs, everything works except with two phones, two Realme 3 pro.
They are crashing the esp.

zekageri, 2020-06-29:
Currently trying to debug the problem. I plugged my phone into the laptop, with usb debugging. Inspecting it with Edge devtools on the laptop. It is crashing when the esp wants to serve a "largeish" file. Its a JS file and its only 200kb.

lorol, 2020-06-29:
I don't have an Android phone handy but I uploaded and downloaded back a 550kb file by PC Mozilla browser - with a File Editor as part of ESPAsyncWebServer  (my tweak and the SmartSwitch application in my clone) it worked well with LITTLEFS.

zekageri, 2020-07-01:
It's still crashing. If i want to send a file wich contains some javascript ( jquery, and font awesome svg (the whole file is 241kb) ) it's just crashing. I dont know. Maybe my 0th core is overloaded and i don't have the memory to send it?

zekageri, 2020-07-01:
Maybe the problem is with the partition. Maybe the async callback can't delay that much time inside it while serving the file.
I've got a `board_build.partitions = large_spiffs_16MB.csv` so the flash is pretty big.

BlueAndi, 2020-07-07:
> * Appreciate if you find the right people to implement it properly in PIO (different command, choice, prompt, py script, config through platformio.ini  per project ... etc.)

Add to *platformio.ini*:
```extra_scripts = replace_fs.py```

Add *replace_fs.py* to project root directory (where platformio.ini is located):
```python
Import("env")
print("Replace MKSPIFFSTOOL with mklittlefs.exe")
env.Replace (MKSPIFFSTOOL = "mklittlefs.exe")
```

Add *mklittlefs.exe* to project root directory as well.

lorol, 2020-07-07:
I will add your notes to  README

BlueAndi, 2020-07-07:
@lorol I got it running on my esp32: https://github.com/BlueAndi/esp-rgb-led-matrix/tree/LittleFS
Did not update SPIFFSEditor.cpp right now, but will in the next step.

lorol, 2020-07-09:
Needed stuff is in gzipped files in /data folder. Let me know which part asks for missing ones.
see https://github.com/lorol/ESPAsyncWebServer/blob/master/src/edit.htm
and  https://github.com/lorol/ESPAsyncWebServer/tree/master/examples/SmartSwitch/data

BlueAndi, 2020-07-09:
I rolled back yesterday ;-)
Updated the editor now by my own (not pushed right now) to show all files in the subdirectories again.

Implemented the following routine (esp32 only), if you are interesting:
```
String SPIFFSEditor::getFilesRecursive(const String& path, bool& isFirst)
{
  String result;
  File dir = _fs.open(path);
  File entry = dir.openNextFile();

  while(entry){
    if (true == isExcluded(_fs, entry.name())) {
      /* Nothing to do */
    } else if ((true == entry.isDirectory()) &&
               (0 != strcmp("/", entry.name()))) {
      result += getFilesRecursive(entry.name(), isFirst);
    } else if (false == entry.isDirectory()) {
      if (true == isFirst) {
        isFirst = false;
      } else {
        result += ",";
      }
      result += "{\"type\":\"";
      result += "file";
      result += "\",\"name\":\"";
      result += String(entry.name());
      result += "\",\"size\":";
      result += String(entry.size());
      result += "}";
    }

    entry.close();
    entry = dir.openNextFile();
  }

  dir.close();

  return result;
}
```

lorol, 2020-07-09:
@BlueAndi thanks for the code shown.
I did the entering to folder by click over directory name, identical for deleting. (right click) with silent tree removal on last file.

lorol, 2020-07-10:
https://github.com/lorol/arduino-esp32fatfs-plugin
You can do FAT partition replacement tests on Arduino, too. Just get this tool, select a part. scheme w/fat and see the examples.
espota has no fat implementation, only use serial port

BlueAndi, 2020-07-14:
@lorol Did you make some measurement regardin read/write speed with LittleFS compared to SPIFFS or FAT?

atanisoft, 2020-07-14:
@BlueAndi https://github.com/RIOT-OS/RIOT/pull/8316 (similar comparison in another project).

and from esp_littlefs itself: https://github.com/joltwallet/esp_littlefs#performance

lorol, 2020-07-14:
@BlueAndi   https://github.com/lorol/LITTLEFS#differences-with-spiffs  - see the table these are just from generic sketch.

lorol, 2020-07-14:
BTW, i combined all 3 uploaders to a single one with drop-down choice. The mk..fs binaries are collected there, too.
See: https://github.com/lorol/arduino-esp32fs-plugin

lorol, 2020-09-05:
Thank you for the feedback. I submitted a request LITTLEFS for esp32 to be reviewed for Arduino Library Manager 
https://github.com/arduino/Arduino/issues/10719 

Closing this.

closed 2020-09-05 by lorol

BlueAndi, 2020-09-05:
@lorol Thanks for sharing. Curious about the IDF integration.

gnalbandian, 2020-09-23:
@BlueAndi are you planning on releasing the tweaked version of spiffseditor that works with LittleFS?

BlueAndi, 2020-09-23:
You will find the version of @lorol here: https://github.com/lorol/ESPAsyncWebServer
Any my one here: https://github.com/BlueAndi/ESPAsyncWebServer/tree/LittleFS