me-no-dev/ESPAsyncWebServer #796: ESPAsyncWebServer crashes when non-existing gzipped FFat file is called

ullix, 2020-06-23:
I lost a file due to FS corruption and noticed this bug. Demo file is attached, this is the ESPAsync server code part:

        server.on("/",      HTTP_GET, [] (AsyncWebServerRequest *request) {request->send(200, "text/html",  "Hello World");});  // ok
        server.on("/test1", HTTP_GET, [] (AsyncWebServerRequest *request) {request->send(FFat, "/test", "text/html");});        // ok
        server.on("/test2", HTTP_GET, [] (AsyncWebServerRequest *request) {                                                     // crash
                                                    AsyncWebServerResponse *response = request->beginResponse(FFat, "/test.gz", "text/html");
                                                                            response->addHeader("Content-Encoding","gzip");
                                                                            request->send(response);
                                                                          });

Neither file 'test' nor file 'test.gz' exists on the FFat file system. When /test1 is called for 'test', the server sends a 404, as it should. But when /test2 is called for 'test.gz', the sketch crashes with this error message:

        Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
        Core 1 register dump:
        PC      : 0x400d10e4  PS      : 0x00060430  A0      : 0x800d6560  A1      : 0x3ffdb540
        A2      : 0x00000000  A3      : 0x3ffc1690  A4      : 0x3ffdd85c  A5      : 0x79a07dab
        A6      : 0x0000000d  A7      : 0x00000000  A8      : 0x800d10e1  A9      : 0x3ffdb520
        A10     : 0x3ffdb550  A11     : 0x3f401087  A12     : 0x79a07dab  A13     : 0x3ffc1690
        A14     : 0x00000000  A15     : 0x3ffdb55c  SAR     : 0x00000018  EXCCAUSE: 0x0000001c
        EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Instead, a 404 should also be send in this situation, and surely no crash should happen.

[BUGDEMO_gzipped_W.ino.zip](https://github.com/me-no-dev/ESPAsyncWebServer/files/4818937/BUGDEMO_gzipped_W.ino.zip)

BlueAndi, 2020-08-24:
Check response for NULL. If the file doesn't exists, it will return NULL IMHO.

ullix, 2020-08-24:
It crashes before you could check.

My point is, whatever the response, it should never crash!

BlueAndi, 2020-08-24:
If request->beginResponse() returns NULL, it will crash by missing NULL check. Anyway, can you decode the backtrace to see really where it happens?

ullix, 2020-08-24:
I am afraid this is beyond my abilities, sorry.
But note that I have included a simple and short demo program in my initial posting.

BlueAndi, 2020-08-24:
Please change initWebServer() function to:
```
void initWebServer() {
    // neither file 'test' nor file 'test.gz' exists on file system
    server.on("/",      HTTP_GET, [] (AsyncWebServerRequest *request) {request->send(200, "text/html",  "Hello World");});  // ok
    server.on("/test1", HTTP_GET, [] (AsyncWebServerRequest *request) {request->send(FFat, "/test", "text/html");});        // ok
    server.on("/test2", HTTP_GET, [] (AsyncWebServerRequest *request) {                                                     // crash
        AsyncWebServerResponse *response = request->beginResponse(FFat, "/test.gz", "text/html");        
                                if (NULL != response) {
                                    response->addHeader("Content-Encoding","gzip");
                                    request->send(response);
                                } else {
                                    request->send(404);
                                }
                              });
    server.begin();
}
```

ullix, 2020-08-25:
Did the change, and indeed, no more crashing, just a plain 404!

But should that response check no be part of the lib?

Pablo2048, 2020-08-25:
> But should that response check no be part of the lib?

No, because It's your fault -you called `response->addHeader()` with the `response` being nullptr. There is nothing that can be done inside the library to avoid this...

Pablo2048, 2020-08-28:
@ullix can you please close this issue?

closed 2020-08-28 by ullix