me-no-dev/ESPAsyncWebServer #800: ESP8266 resets on POST request

jazibdawre, 2020-06-27:
The request handler for my POST callback looks like this
```
void post_settings(AsyncWebServerRequest *request, uint8_t *data, size_t len, size_t index, size_t total) {

  if(!index) {    //start of data
    req_json = (char*)data;
  } else {    //midway
    strcat(req_json, (char*)data);
  }

  if(index + len == total) {    //End data
    const size_t capacity = JSON_OBJECT_SIZE(5) + 50;
    DynamicJsonDocument doc(capacity);
    deserializeJson(doc, req_json);
    //use json to set variables

    AsyncWebServerResponse *response = request->beginResponse(200);
    request->send(response);
  }
}
```

A version of this using the ESP8266WebServer library runs properly, however when I use the the async server the esp resets.
The esp prints this on serial `ets Jan 8 2013,rst cause:2, boot mode:(3,6)`. From what I've gathered this is due to the rst pin and not the s/w or h/w watchdogs. It was said to add yield() to the code so that the esp can do it's thing but I can't put it inside of the callback. Should I put it inside my arduino loop()?
Any help would be appreciated. Thanks

BlueAndi, 2020-06-27:
The function parameter "data" is not be a terminated string. Therefore the length is given by the function parameter "len".

jazibdawre, 2020-06-28:
it still resets after adding "\0" at end.

BlueAndi, 2020-06-28:
How does req_json looks like? It looks for me, its just a pointer which takes in the first iteration the data pointer. And in the 2nd iteration it overwrites the original data buffer.
And how did you terminate? If you add a '\0' to the data buffer end, you may overwrite the buffer too.

jazibdawre, 2020-06-28:
req_json is of char* type. Basically in every iteration it appends the data (srtcat) until the final piece, then it parses it. The parsing stuff is working correctly even though the esp restarts. Sometimes it does that without restarting too.

BlueAndi, 2020-06-29:
You use the buffer of the data pointer, you get in the first call. But you are not the owner of this buffer, which will be released (IMHO its on the heap) after leaving the callback, isn't it? The next calls overwrite more and more on the heap. It may work sometimes and sometimes not.

jazibdawre, 2020-07-01:
Ah, Should I change req_json to string and append the data to it or is there a better way? Thanks for your help

closed 2020-07-03 by jazibdawre