me-no-dev/ESPAsyncWebServer #802: ESPAsyncWebServer sends Connection: close header yet fails to close TCP/IP connection server-side

avillacis, 2020-07-02:
Using ESPAsyncWebServer at master tip f13685ee97675be2ac9502d177d3024ebc49c1e0 and AsyncTCP at master tip ca8ac5f919d02bea07b474531981ddbfd64de97c on an ESP32.

I was trying to study a different issue (too many simultaneous SPIFFS requests trigger async_tcp watchdog) by using the "ab" benchmarking program from Apache Web Server. I then realized that the "ab" program depends on the webserver actually closing the TCP/IP socket when requesting an URL and seeing the Connection: close header. Instead, the ESP32 webserver sends the data, including the Connection: close header, and leaves the socket open until the client times out.

To test:

ab -v 4 http://{ESP32_SERVER_IP}/someresource

Expected: ESP web server sends response with Connection: close header, then **actually** closes the socket

Actual: ESP web server sends response with Connection: close , and then leaves the socket open, as if it were to transmit more data.

Besides this issue obstructing benchmarking and diagnostics, this could lead to a DOS server-side if the socket itself, and possibly the Response object are not freed because the socket is not being closed.

This issue is apparently not affected by the type of web handler that serves the response. Static files are affected, as well as JSON endpoints that invoke application code. Also, 404 handlers are also affected.

avillacis, 2020-09-01:
Not stale. Not fixed either.

avillacis, 2020-11-02:
Not stale. Still not fixed either.

ramarro123, 2020-12-08:
@me-no-dev i am experiencing this issue as well, unfortunatley some of my esp32 are reachable by outside and even without "bad intention" it's very easy sto stop them with an AB-like things (ie: a client that doesn't close the connection on his side)

i checked out the merged version and thidn't change anything

x-labz, 2021-01-16:
@me-no-dev 
I can confirm the issue as well. There is some mysterious problem how the ESPAsyncWebServer is NOT closing the connections properly. 
If the request issued from a PC browser, everithing looks correct, the PC browser doesn't complain about any issue.
BUT if the request is sent from an ESP8266 board - eigther running c++ ESPAsyncTCP client or micropython - the request is stalled, the client is waiting forever ( or at least for the timeout period, if implemented )

avillacis, 2021-03-19:
Still not solved so far.

maxelle, 2021-03-22:
Still not solved, ESP32 unusable with mobile devices

avillacis, 2021-06-02:
Still not solved so far...........

avillacis, 2021-08-03:
Still not solved.

avillacis, 2021-10-12:
Still not solved.

closed 2022-04-16 by stale[bot]

jcelerier, 2024-10-09:
I still seem to see this behaviour. My code is:

     server.on("/", HTTP_GET, [] {  request->send(200, "text/json", "{ some json }"); });

but from the client I never see an EOF.

PR #836 (2025-01-20): Explicitly close socket after response has entered finished state
In at least two places in the code, the response object is evaluated
on whether it has finished sending data to the client. If the response
has already finished, or enters the finished state after sending data,
the underlying socket should not be left open. Failure to close the
socket in these scenarios cause the remote client to timeout expecting
more data unless it closes the connection on its own.

Fixes: #802