me-no-dev/ESPAsyncWebServer #804: Advice using websockets

esp32noob, 2020-07-05:
Thank you for looking at this.

I'm trying to get to grips with websockets but I'm missing something fundamental.  I have condensed down all the examples I've come across to the code below.  My understanding is if a websocket event occurs then the serial.println line will execute in the terminal, but it doesn't. I do get a blank response with no errors either from the web page or the terminal.  Obviously I've missed something, could someone please point out my error.

`#include <Arduino.h>
#include <ESPAsyncWebServer.h>

AsyncWebServer server(80);
AsyncWebSocket WebSocket("/ws");
//
void onWsEvent(AsyncWebSocket *server, AsyncWebSocketClient *client, AwsEventType type, void *arg, uint8_t *data, size_t len)
{
  Serial.println("websocket request");
}
void setup()
{
  Serial.begin(115200);
  WiFi.begin("SSID", "PASSWORD");
  vTaskDelay(pdMS_TO_TICKS(1000));
  Serial.println(WiFi.localIP());
  //
  WebSocket.onEvent(onWsEvent);
  server.addHandler(&WebSocket);
  server.begin();
}
void loop()
{
  WebSocket.cleanupClients();
  vTaskDelay(pdMS_TO_TICKS(100));
}`

zekageri, 2020-07-08:
### It works for me like this:

```
AsyncWebSocketClient * globalClient = NULL;
void onWsEvent(AsyncWebSocket * server, AsyncWebSocketClient * client, AwsEventType type, void * arg, uint8_t *data, size_t len){
  if(type == WS_EVT_CONNECT){
    Serial.println("Async_WebSocket_Connected");
    globalClient = client;
  } else if(type == WS_EVT_DISCONNECT){
    globalClient = NULL;
    Serial.println("Async_WebSocket_Disconnected");
  }
}
```

**How do you want to connect to it? In javascript? can you show me that code?**

BlueAndi, 2020-07-08:
I just followed the example here: https://github.com/me-no-dev/ESPAsyncWebServer#async-websocket-plugin

closed 2020-07-08 by esp32noob

esp32noob, 2020-07-08:
Thank you zekageri and BlueAndi, this has helped me enormously and has enabled me to move on and continue my learning.  As always the me-no-dev example was excellent but the problem was I didn't have enough knowledge to put it into context.  I also found https://m1cr0lab-esp32.github.io/remote-control-with-websocket/ to be very helpful.

I've made an esp32 variation of https://create.arduino.cc/projecthub/gbarbarov/open-led-race-a0331a and want to present a realtime time update on a web page and believe that websockets and javascript will allow me to do this.  Thanks again.