me-no-dev/ESPAsyncWebServer #811: AsyncWebSocket::cleanupClients from main loop has a race condition causing a crash

Dmytry, 2020-07-26:
The crash happens in AsyncWebSocket::count() , in the lambda function it passes to count_if.

I suspect this is because removal of a disconnected client from the list happens during AsyncWebSocket::count(). 

I seem to have fixed it by calling _server->cleanupClients(); from AsyncWebSocketResponse::_ack (prior to new client creation), that seems to resolve the crash and also eliminate the need for polling. 

I'm not sure if that is a good solution and if _ack might be called simultaneously on several threads, though.

closed 2020-10-09 by stale[bot]

Belleson, 2024-03-22:
I had a similar problem with unhandled exception AsyncWebSocket::cleanupClients(unsigned short) at.pio/libdeps/devkit/ESP Async WebServer/src/AsyncWebSocket.cpp:924

and made the source code change suggested by @Dmytry. It's early but seems to have solved it, thanks!

softwarecrash, 2024-04-09:
> I had a similar problem with unhandled exception AsyncWebSocket::cleanupClients(unsigned short) at.pio/libdeps/devkit/ESP Async WebServer/src/AsyncWebSocket.cpp:924
> 
> and made the source code change suggested by @Dmytry. It's early but seems to have solved it, thanks!

hey, i have the same issue, can you provide the changes that must made or a short example?
thank you

softwarecrash, 2024-04-09:
> in AsyncWebSocket.cpp, change the last function in the class, "_ack" to
> 
> ```
> size_t AsyncWebSocketResponse::_ack(AsyncWebServerRequest *request, size_t len, uint32_t time){
>   (void)time;
>   _server->cleanupClients(); // ADD THIS see https://github.com/me-no-dev/ESPAsyncWebServer/issues/811
>   if(len){
>     new AsyncWebSocketClient(request, _server);
>   }
>   return 0;
> }
> ```
> 
> Then in your app's main.cpp, remove socket.cleanupClients(); from loop() { . . . }
> 
> Remember that if you refresh the library, it will overwrite changes you've made (although it hasn't been updated in a while).

thats fast, thank  you verry much, i will try it