me-no-dev/ESPAsyncWebServer #816: spiffs download works only for small files ?

pvdw0310, 2020-08-01:
Hi I like to download some files which are saved on a esp8266 with LittleFs. I use =>

`             request->send(LittleFS, uplfile, String(), true);    `

If the file is smaller then about 1930 bytes then I can download the total file. If the file is bigger, then the download stays hanging at about 1930 bytes.

How can I solve that problem?
Many thanks

mntolia, 2020-08-02:
Send the response in chunks.

pvdw0310, 2020-08-02:
how can I download the file in chunks?

gnalbandian, 2020-08-14:
I have found that having multiple file requests doesn't work well.
It's better to make just one file with all the html content and javacript as well.

pvdw0310, 2020-08-14:
I only select 1 file with the filename in uplfile. If that file is bigger the about 1930bytes then de download hangs. If that file is smaller then it works fine. Is there a memory problem or can I make a buffer bigger (witch buffer)? Or how can I cut the file in blocks to download from SPIFFS to pc.

closed 2020-11-05 by stale[bot]