me-no-dev/ESPAsyncWebServer #817: Server unreachable when esp doing anything else.

TomaszKisiel, 2020-08-02:
When I do anything more complicated than one line of code, my server becomes unreachable.

My base code look like this:

    #include <ESP8266WiFi.h>
    #include <ESPAsyncTCP.h>
    #include <ESPAsyncWebServer.h>
    #include <LittleFS.h>
    #include <Wire.h>

    AsyncWebServer server(80);

    void setup( void ) {
        if(!LittleFS.begin()){
            Serial.println("An Error has occurred while mounting SPIFFS");
            return;
        }

        WiFi.begin( "ssid", "password" );
        while (WiFi.status() != WL_CONNECTED) {
            delay(100);
        }

        server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
            // request->send(200, "text/plain", "Hello, sensor: ");
            if ( !request->authenticate( accessPointSSID, accessPointPassword ) ) {
                return request->requestAuthentication();
            }
            request->send(LittleFS, "/index.html", String(), false, htmlTemplate);
        });
        server.on("/style.min.css", HTTP_GET, [](AsyncWebServerRequest *request){
            request->send(LittleFS, "/style.min.css", "text/css");
        });
        server.begin();
    }

When my loop function look like this:

    void loop( void ) {}

everything works just perfect. But when i try to use e.g lcd display

    void loop( void ) {
        lcd.clear();
        lcd.setCoursor( 0, 1 );
        lcd.print( "Doesn't work :(" );
        delay( 500 );
    }

server becomes unreachable forever. Sometimes ESP restarts without any reasons. 

Is there anything I should do to make this work or maybe this library only works when esp is doing nothing else?

TomaszKisiel, 2020-08-09:
.

ferranprojects, 2020-08-28:
Same issue...waiting for an answer...

athusrv, 2020-09-17:
avoid using delay. Prefer doing something like:

```
unsigned long timeout = 0;

void loop() {
   if(millis() > timeout) {
      // do your stuff

     timeout = millis() + 500;
   }
}
```

Pablo2048, 2020-09-17:
Agree with avoiding the delay, but never do it this way - it does not work in case of millis()+... rollover. Use this:
```cpp
uint32_t last = 0;

void loop(void) {
  if ((millis() - last) > 500) {
    last = millis();
    // do your stuff

  }
}
```
instead...

TomaszKisiel, 2020-09-25:
Why should I avoid delay ? What is wrong with this function ?

lorol, 2020-09-25:
The delay should be avoided in server handler functions on.xxx because the async queues and sends out stacks automatically.
Here, I believe the problem is caused by wire low-level functions or lcd driver, which can block server process same way.

zekageri, 2020-11-24:
> The delay should be avoided in server handler functions on.xxx because the async queues and sends out stacks automatically. If you hold too long, the things get overfill and/or WDT reacts.
> Here, I believe the problem is caused by wire low-level functions or lcd driver, which can block server process same way.

Async server using the core 0 on its own tasks isn't it? 
How can a delay(something) in the core 1 inside the loop affect it?

Pablo2048, 2020-11-24:
> How can a delay(something) in the core 1 inside the loop affect it?

The original issue was about ESP8266 so no core 0/1 here. But anyway when the ESP restarts it spits out the reason for the restart. Until anyone catch this info (and try to decode it with exception decoder) we are unable to help/explain it IMHO.

closed 2021-02-07 by stale[bot]