me-no-dev/ESPAsyncWebServer #818: AsyncWebSocketClient  client->printf

ats3788, 2020-08-03:
Hello I have 
an `void onWinSocketEvent(AsyncWebSocket * server, AsyncWebSocketClient * client, AwsEventType type, void * arg, uint8_t *data, size_t len)
{
  if(type == WS_EVT_CONNECT){

    Serial.println("Websocket client connection received");

  } else if(type == WS_EVT_DISCONNECT){
    Serial.println("Client disconnected");

  } else if(type == WS_EVT_DATA){

    Serial.println("Data received: ");

    for(int i=0; i < len; i++) 
      Serial.write(data[i]);
     Serial.println();
  }

client->printf("Hallo Daisy %d" , millis());

}

..........................
 ws.onEvent(onWinSocketEvent);
 server.addHandler(&ws);
..................
`

and java Script
`

let url = "ws://192.168.1.250/radio_ws"

let socket = new WebSocket(url);

socket.onopen = function(e) {
  alert("[open] Connection established");
  alert("Sending to server");
  socket.send("My name is John");
};

socket.onmessage = function(event) {
  alert(`[message] Data received from server: ${event.data}`);
};

socket.onclose = function(event) {
  if (event.wasClean) {
    alert(`[close] Connection closed cleanly, code=${event.code} reason=${event.reason}`);
  } else {
    // e.g. server process killed or network down
    // event.code is usually 1006 in this case
    alert('[close] Connection died');
  }
};

socket.onerror = function(error) {
  alert(`[error] ${error.message}`);
};

// show message in div#messages
function showMessage(message) {
  let messageElem = document.createElement('div');
  messageElem.textContent = message;
  document.getElementById('messages').prepend(messageElem);
}  `
this works great, but I would like to send every couple of seconds infos to the webpage like this
client->printf("Hallo Daisy %d" , millis()); but all what I tried did not work. Any clue how

BlueAndi, 2020-08-03:
https://github.com/me-no-dev/ESPAsyncWebServer#methods-for-sending-data-to-a-socket-client

ats3788, 2020-08-04:
Thank you BlueAndi have to check it out, yes very helpful
but still ws.printf((uint32_t)client_id, arguments...);  how do I get the Client ID
How does this work  ws.getClients();  
can't find an example

```
 ws.printfAll("<- %d ->" , rand());
  ws.cleanupClients();
```
This doesn't or better what number I have to use
```
 ws.printf(number, "<- %d ->" , rand());
 ws.cleanupClients(
```);

BlueAndi, 2020-08-04:
You will find the example how to get the client id in the readme as well, see
https://github.com/me-no-dev/ESPAsyncWebServer#async-websocket-event

Have a look at the place where WS_EVT_CONNECT is considered.

ats3788, 2020-08-11:
THX BlueAndi I got it
void onWsEvent(AsyncWebSocket * server, AsyncWebSocketClient * client, AwsEventType type, void * arg, uint8_t *data, size_t len)
{
  if(type == WS_EVT_CONNECT){
    Serial.printf("ws[%s][%u] connect\n", server->url(), client->id());
    client->printf("***> Hello Client %u :)", client->id());
     ClientID = client->id();

and 

  ws.printf(ClientID, "Hallo NeoPixel");
       ws.cleanupClients();

Works perfect now

closed 2020-10-24 by stale[bot]