me-no-dev/ESPAsyncWebServer #819: WEIRD PROB / Can't send string that's 11 characters or more.

tsotnek, 2020-08-05:
Hello I have major problem when I try to send any string that is 11 characters or longer to the server it doesn't get displayed on the webserver, it just gets weird font like **L1�?** or something like that. This is the simple code I am using on the server.

`server.on("/", HTTP_GET, [] (AsyncWebServerRequest *request) {

    AsyncResponseStream *response = request->beginResponseStream("text/html");
    String elevencharacter = "eleven character";
    response->print("<!DOCTYPE html><html><body>
    response->printf("<p>%s\r\n<br></body></html>", elevencharacter);
    request->send(response);`

Any help is appreciated on what could cause a string to not display correctly.

alvahdean, 2020-08-06:
I'm not 100% on the response->printf method signatures but I do know that
the system printf requires a char* so you can't pass a String directly. You
need to call the method to get the C string for the String object and pass
that in printf. I forget the method name but I think it's something like
cstr() or tocstr(). This very thing has bitten me more than once so I hope
that helps.

-Dean

Pablo2048, 2020-08-06:
The `%s` printf require `const char *` value so use it like this:
```c
response->printf("<p>%s\r\n<br></body></html>", elevencharacter.c_str());
```

tsotnek, 2020-08-06:
By the gods, it solved the problem!!! It bugged be quite some time... I thank you both greatly.

closed 2020-08-06 by tsotnek